ikarus23/MifareClassicTool #279: Samsung Galaxy S9

MarcusWolschon, 2019-12-13:
You list the Samsung Galaxy S9 as compatible but it has a Samsung NFC chipset and not an NXP chipset.
The Samsung Galaxy S9 I have here clearly does not detect Mifare Classic or Mifare Desfire EV1 cards.  (Tested with the Ikarus Mifare Classic Tool, the NXP Taginfo app and using the Android API myself.)

ikarus23, 2019-12-14:
Interesting, this is the first time I hear of an Samsung Galaxy S9 not working with Mifare Classic. Had it worked before? Some users reported their Samsung (Note 8) devices have stopped working after an update. But this was a year ago.

But what is the weirdest part is your observation about Mifare DESFire EV1. I'm used stories about Mifare Classic not working, but Mifare DESFire is different and should work. What exactly did you witness? Was there an error? Was the tag even detected? Were any other tag detected (e.g. you credit card)?

MarcusWolschon, 2019-12-16:
Both tags are not even detected.
Not a single line about it in the logcat either.
Tested a Mifare Classic and a Mifare Desfire EV1 with 4K of memory.
It may of cause have worked in an earlier firmware.
Both tags work on a different phone (Blackview BV9600 Pro) and other tags work just fine on the Samsung Galaxy S9.

MarcusWolschon, 2019-12-16:
Android 9
Firmware Build number
PPR1.180610.011.G960FXXS7CSK4
Kernel
4.9.59-16947752

ikarus23, 2019-12-16:
The same tags working on another phone and different tags working on the S9 makes everything even stranger. What other tags are working on the S9?

MarcusWolschon, 2019-12-17:
Texas Instruments NfcV  (-3 and -2 compatible) works fine
Infinion  NfcV  (-3 and -2 compatible) works fine

I just tried 4 other Mifare Classic cards to make sure. None of them where detected at all on the Samsung.
Two Infinion ISO14443-4 payment cards where also not detected (running Austria Card ACOS and G&D SmartCafe).
All worked fine on different phones.
Tried NFC Method "Android OS", "embedded secure element", "SE SIM card" and "Autodetect" in the Samsung settings.

ikarus23, 2019-12-17:
That is what I expected. It seams you device is not able of reading ISO 14443 tags, but only ISO 15693 (NfcV) tags. This is not normal. There must be something wrong with the Android ROM, the driver or, more likely, your hardware. I don't think there is much I can do. Sorry.

ikarus23, 2020-01-24:
Any news on this? Have you found the issue? Can we close this?

MarcusWolschon, 2020-01-27:
No answer from Samsung if this is an intended firmware change.
Since it's a software protocol and all other cards work.We can rule our a hardware defect.

ikarus23, 2020-01-30:
If it is not hardware, this is even stranger. I never heard of an S9 owner with the same issue. The S9 I'm having access to, has no issues with Mifare Classic or other tags.

ikarus23, 2020-08-15:
I will close this for now. Feel free to report back if you learned something new about this issue.

closed 2020-08-15 by ikarus23