ikarus23/MifareClassicTool #281: MCT on Huawei App Gallery

dglance, 2020-01-09:
Could you upload MCT for distribution through the Huawei App Gallery - The US has sanctioned Huawei and the new phones including the Mate 30 Pro (which works fine with MCT) ship without access to Google Play. 

I believe it is relatively straightforward to add the APK to their App Gallery. 

If you don't have the time or effort - can I do this on your behalf?

ikarus23, 2020-01-10:
Fair enough, but it comes down to pretty much this:
> If you don't have the time or effort - can I do this on your behalf?

I'm sorry but I don't want to invest the extra time for another app store. Is F-Droid a thing for Huawei users? Because the app is available there as well. However, if you are willing to do it, feel free to publish it in Huawei's App Gallery. But don't forget: it's a responsibility too. You have to keep it up to date and there will be questions asking for support. But I guess you can redirect them to this github repo my mail from the Google Play store.

Please use the signed APKs from https://www.icaria.de/mct/releases/ and copy the store entry text from Google Play. Thank you very much for helping to distribute this app!

dglance, 2020-02-23:
Thanks. Slight delay in getting accounts set up - it requires id verification etc - will let you know when done.

w2016561536, 2020-02-28:
First , I'm a Chinese
The HUAWEI MATE 30 can't install play store have been solved
You can check it on HUAWEI Community https://cn.ui.vmall.com/thread-21785219-1-1.html

dglance, 2020-03-02:
@ikarus23 the app is now on the Huawei AppGallery - don't know if you want to add this to the readme?

@w2016561536 - thanks for the tip, you can also use LZ Play (which is still available and I have installed and it works fine) - however, it is also good that you can get it directly now from the Huawei AppGallery.

ikarus23, 2020-08-15:
I've added the link to the readme. Thanks for helping out with this.

closed 2020-08-15 by ikarus23