ikarus23/MifareClassicTool #288: How to Have a dump file with 1024 Bytes

frem95, 2020-02-09:
Hello,
Old french man here. I nerver learn Englich at School.
I never use a forum.
I have on my phone MCT V 2.3.1
Mct create a 2293 Bytes files
I read a Mifare 1 Ko
How to export dump file with 1024 Bytes ?
I do Compare dump file create with Mct and same Dump File create with Kali Linux 2017
With Kali i use : mfoc -f   extended-std.keys -O MyBadge.dmp (1 Ko)
Badge is Product by Cogelec with 2  Trade Mark : Hexact or Intratone For Bulding Access
MCT can't write good Clone ( errors :Write Sectors and Error Date)
No Probleme with Kali to Write on Chinese  Card : Bloc 0 Rewritable
My target : Create a 1024 bytes file (like Kali)  since 2293 bytes (MCT)
Thank You

ikarus23, 2020-02-14:
Hi, your English is just fine ;)
Yes, MCT dump files have a different format. I've started to implement a converter to export compatible file formats. But is is not finished yet. But you can use the python tools at https://github.com/ikarus23/MifareClassicTool/tree/master/tools/dump-file-converter. Use mct2eml.py and then eml2mfd.py. `.mfd` files are the raw 1024 byte files. You can also convert the .bin/.mfd file to a MCT compatible file (mfd2eml.py --> eml2mct.py). MCT has a compare tool to easily diff dumps.

frem95, 2020-02-14:
Hi,
Thanks for the rest but I don't know python, I no longer use MS Office
With help on a forum, we got a convertion
from 2293 bytes to 1024 bytes in a few lines of VBA in Excel
Here is the code:

Sub test ()
    Dim s As String, d As String
    s = "F: \ Electronique \ DataRecupKali \ DataRecupHuaweiMct \ dump-files \ badgebleuarduiUID_F5459379"
    d = "F: \ Electronique \ DataRecupKali \ DataRecupHuaweiMct \ dump-files \ ConvertMct2Kali \ badgebleuarduiUID_F5459379.dmp"
    convert s, d
End Sub

Sub convert (fileSource As String, fileDest As String) 'path + file
    Dim strLine As String, strContent As String, tmp
    Dim numfich As Integer, i As Long, j As Long, k As Long
    
    numfich = FreeFile
    Open file Source For Input As #numfich
    Do While Not EOF (numfich)
        Line Input #numfich, strLine
        tmp = Split (strLine, vbLf)
        For i = 0 To UBound (tmp)
            If Left (tmp (i), 1) <> "+" Then
                For j = 1 To Len (tmp (i)) Step 2
                    strContent = strContent & Chr ("& H" & Mid (tmp (i), j, 2))
                Next j
            End If
        Next i
    Loop
    Close #numfich
    numfich = FreeFile
    Open fileDest For Output As #numfich
    Print #numfich, strContent;
    Close #numfich
    Beep
End Sub

It should be easily convertible to python
[BeepIntratone4BtnNeuf.zip](https://github.com/ikarus23/MifareClassicTool/files/4207265/BeepIntratone4BtnNeuf.zip)

closed 2020-02-14 by frem95

frem95, 2020-02-14:
Hi,

		

Thanks for the rest but I don't know python, I no longer use MS Office

With help on a forum, we got a convertion

from 2293 bytes to 1024 bytes in a few lines of VBA in Excel

Here is the code:

	
			

Sub test ()

		

    Dim s As String, d As String

    s = "F: \ Electronique \ DataRecupKali \ DataRecupHuaweiMct \ dump-files \ badgebleuarduiUID_F5459379"

    d = "F: \ Electronique \ DataRecupKali \ DataRecupHuaweiMct \ dump-files \ ConvertMct2Kali \ badgebleuarduiUID_F5459379.dmp"

    convert s, d

	

End Sub

		
			

Sub convert (fileSource As String, fileDest As String) 'path + file

    Dim strLine As String, strContent As String, tmp

    Dim numfich As Integer, i As Long, j As Long, k As Long

    

		

    numfich = FreeFile

	

    Open file Source For Input As #numfich

    Do While Not EOF (numfich)

        Line Input #numfich, strLine

        tmp = Split (strLine, vbLf)

        For i = 0 To UBound (tmp)

            If Left (tmp (i), 1) <> "+" Then

                For j = 1 To Len (tmp (i)) Step 2

                    strContent = strContent & Chr ("& H" & Mid (tmp (i), j, 2))

                Next j

	

            End If

	

        Next i

		

    Loop

		

    Close #numfich

	

    numfich = FreeFile

	

    Open fileDest For Output As #numfich

    Print #numfich, strContent;

    Close #numfich

	

    Beep

		

End Sub

		
			

It should be easily convertible to python

			

Do you know these badges, attached files of a new badge?