ikarus23/MifareClassicTool #290: Keys not found or dead sector

terragady, 2020-02-26:
Keys not found or dead sector

I have few sectors with this message, after cloning the result to other tag it does not work so I assume there is something hidden in those sectors. Can it be read someway? Is it a different key? Or 2 keys?

ikarus23, 2020-02-26:
Yes, you can read the data from all sectors if you have the keys. But in any case, you need the keys! Have a look at https://github.com/ikarus23/MifareClassicTool#getting-started and read the Mifare Classic datasheet.

terragady, 2020-02-26:
Oh OK so it just didn't match the correct key (not in the dB you provided). I assumed they are just empty as the other ones are but maybe not. So I guess it is then not possible to do it with just phone or there is any trick or other app for that? Or I can use phone as an nfc reader for my PC?

ikarus23, 2020-02-26:
Yes, it is most likely that the interesting data is stored in the sectors to which you do not have the key. And no, I don't know of any way to use your phone to crack the keys.

terragady, 2020-02-26:
Yeah that's what I thought. I will search the net. I am also the owner of that but I have no idea where to get the keys or information and they want some terrible fee at the service to make more copies.

ikarus23, 2020-02-26:
You should be fine with a cheep ACR122u USB reader. The rest is software. Most of it is already available within the Kali Linux distribution.

If they check the UID of the card as well, you need special tags, which allow to change the UID as well.

ikarus23, 2020-08-15:
I will close this now. Feel free to ask more questions if there are any.

closed 2020-08-15 by ikarus23