ikarus23/MifareClassicTool #291: 我用mct添加一些已知密钥时，它显示不是十六进制！！

lcq666, 2020-03-16:
[image]

以下是我添加的密钥：
7E3D5A01B5E0
2C6E99455E3A
49950410B212
20FD378DBE45
D168DEE8ADED
FFFFFFFFFFFF
6013E2FA096B

我希望添加已知的密钥能解除这个限制！

ikarus23, 2020-03-16:
Not sure what you want. I do not speak Chinese. Do you want me to add the keys? What do you know about the keys? Are they unique to a card? Or to an installation? Are they the same on all cards of the same product?

lcq666, 2020-03-16:
When I used this software to add my known key, it made an error, indicating that the key is invalid, not hexadecimal number, can this limit be removed? I added the following key, and then there was an error!

7E3D5A01B5E0
2C6E99455E3A
49950410B212
20FD378DBE45
D168DEE8ADED
FFFFFFFFFFFF
6013E2FA096B

When I add the key of appeal, it will prompt me the problem of appeal. I hope this restriction can be lifted when I add the key. Thank you!

ikarus23, 2020-03-16:
Interesting, the look OK to me. Are you sure the lines are correct? No whitespaces at the end? Did you paste them from a non-ASCII source? A key should be 12 hexadecimal characters.

ikarus23, 2020-03-16:
I copied the keys to my phone and everything worked.

lcq666, 2020-03-17:
Awkwardness, I before in the old key file that oneself build new add a part of key, when save again appear error, then just now I directly create a key file paste in, everything is normal, really strange, sorry, disturb you

lcq666, 2020-03-17:
This software can add a card simulation function is more perfect.

ikarus23, 2020-03-17:
Thanks for the update. And no, simulating Mifare Classic cards with Android is not possible.

closed 2020-03-17 by ikarus23