ikarus23/MifareClassicTool #292: INFO: Public library wrapper around the MCT application released

maxieds, 2020-03-22:
Hello. This is just a friendly announcement related to the wonderful, and extremely useful, MCT Android application. This seemed like the best place for the information to go.

I just wanted to make everyone who might be interested in this aware that a new library based on the MCT codebase has been released: [Mifare Classic Tool Library](https://github.com/maxieds/ParklinkMifareClassicToolLibrary/releases/tag/1.1.8). It was developed by @maxieds as freelance software work for *Parklink LLC*. The code is completely open source. 

There is also a demo application that provides a detailed working example of howto integrate the library into custom Android / Java code: [MCT Library Demo](https://github.com/maxieds/MCTLibraryDemo/releases/tag/1.1.9). The demo application is also available for installation and testing on [Google Play Store](https://play.google.com/store/apps/details?id=com.maxieds.ParklinkMCTLibraryDemo).

Please use the [library repo issues pages](https://github.com/maxieds/ParklinkMifareClassicToolLibrary/issues) to give me feedback on the library, and/or to post announcements about usages of this library. I'd like to stay in the loop if users find this library, or extension of MCT, useful.

ikarus23, 2020-03-25:
Interesting! Never thought the code of this project would make it to a library. Also I'm not sure if this is a good idea, because the code is not that good ;) But if it helps somebody, why not? :)

Just out of curiosity, why does Parklink LLC need this functionality?

maxieds, 2020-03-25:
@ikarus23 
Technically I didn't sign a NDA, though to remain professional, I will be brief: They, I believe, need the information that the MCT can glean, but not within that app format, e.g., they need to snarf the contents of a MFC tag or interact with one using preset keys, then use the resulting data to do other things. 

It would have made no sense to write this code completely from datasheet specs from a time nor practical perspective. So your app was the benchmark. The library was always meant to be open source in accordance with the fact that your code, which is the bulk of the library, is open source software. It got put on my back burner for a while with school work being my priority for just a while longer. Now I need to do an equivalent of the MF32Crack routines in [my app for the Chameleon Mini](https://github.com/maxieds/ChameleonMiniLiveDebugger), so I made the repo public. 

How do you like the tester application? Pretty slick interface, right?

ikarus23, 2020-08-15:
Lets see where this library goes. Would be nice to find it in some open source projects.

I'm closing this now since I'm cleaning up the issue tracker ;)

closed 2020-08-15 by ikarus23