ikarus23/MifareClassicTool #293: Some alphabets incorrectly read from block of Mifare Classic 1K Tags

iamporus, 2020-03-26:
I am trying to write the following string into a block of Mifare Classic 1K card.
"abcdefghijklmnop"
The write operation completes successfully but on reading the same block, I'm seeing the following hex dump
"616263646566676869FFFFFFFFFFFF70"
Why I am losing letters from 'j' to 'o?'

Please help me understand if I'm missing anything.

ikarus23, 2020-03-26:
This is very strange! What block are you trying to write to? And how do translate the string to hex? Does the data you want tor write look like this when you enter it? `6162636465666768696a6b6c6d6e6f70`

iamporus, 2020-03-27:
Sorry to bother you. I was using a different logic to convert hex string to byte array which was causing the problem. I switched to your implementation and it started showing me the correct letters.
Closing the issue.
Thanks.

closed 2020-03-27 by iamporus

ikarus23, 2020-03-27:
No worries, and thanks for the feedback.