ikarus23/MifareClassicTool #295: randomize uid

alfs, 2020-04-22:
For CUID cards, it would be valuable to have a simple UID change/edit/randomize function (which would also recalculate BCC of course).

For example in the Select dump / advanced options / Enable writing of the manufacturer block,
there could be a box to enable UID editing and randomization:
```
[X] Advanced: enable writing to manufacturer block
     AABBCCDD  [Set UID] [Randomize UID] [Undo]
```

Or it could be a separate (O) option below "Incr/Decr value block" option, e.g.

(O) Advanced: UID editor

Both places have merits, I think. The first would be a "dump config" style, while the second would be a more "post-write editor" style.

The first should be very straight-forward to implement, I guess. Just rewrite the dump contents, calculate bcc and write as usual.

ikarus23, 2020-05-02:
Hmm, there is something close to it. It's under Tools --> Clone UID. Only the "random" part is missing. You have to enter or scan a UID. I guess it would be easy to add an additional button to fill the text filed with a random UID. Of course the tool needs to be renamed to something like "Clone or Randomize UID".

ikarus23, 2020-05-02:
Just out of curiosity: Are there many use cases where a randomized UID is needed? All typical hacking and cloning cases that come to my mind need a specific UID (e.g. the UID of the original tag).

alfs, 2020-05-03:
Aha, I was not aware of this menu, thanks. It would be the natural place for a randomizing function.

Two use-cases:
- Production/printing of own cards, where you may want to use CUID cards for some reason. I.e. stamp out cards with same sectors, but with unique id's.
- Obfuscation e.g. for stored-credit, where the uid is irrelevant, but may be logged. Summing the total credits per uid would reveal abuse. So in pen-testing, it would be useful to test uid randomization escape abuse detection, for systems that do not tie credit to the uid.

ikarus23, 2020-08-15:
Added in ff91516f45739b138e5984d61d8e8ef3b30cc3ba via "Clone UID Tool".

closed 2020-08-15 by ikarus23