ikarus23/MifareClassicTool #296: successive readings do not give the same dump

akhodaba, 2020-05-23:
Hi there,

I did few successive readings on a badge and compared the dumps with the diff tool. The readings differ.
On another tag, the reading takes a long time and can not read some sectors. Here as well successive reading differ and each time a different set of sectors are not present.

What this may mean?
Many thanks for your help.

ikarus23, 2020-05-25:
Hi! What device are you using? There are some models out there (e.g. Nexus 5X) which have an bug causing inconsistent reads. So fare there is nothing I can do because it is an Android issue. However, there is a workaround in MCT. Try the "Retry authentication" option with values like 5 or 10. This should give you more consistent readings (but will slow down the mapping process).

akhodaba, 2020-05-26:
Thanks for the update. My device is Redmi Note 8 Pro, with Android 9. I tried with Retry option set to 10. In fact the reading was not slower and the inconsistencies persists. Do you know of a similar bug than the Nexus on the Redmi?

ikarus23, 2020-05-26:
Well, this is really strange. What is inconsistent? Single bits and bytes? Whole blocks or sectors?

akhodaba, 2020-05-26:
I did not notice a sector difference. But a missing sector yes.
I send you 2 diffs of the same tag bewteen 3 dumps.

[image]

[image]

ikarus23, 2020-05-26:
Wow, ok, there is something really wrong here! Never saw this before. This really looks like a broken NFC driver or something like it. Could you please cross check with other apps like NXP's TagInfo? Do they show the same behavior?

akhodaba, 2020-05-27:
Hm! I did various tests:

With the above tag:
- MCT on another Redmi Note 8 Pro, with Android 9: same behavior
- NXP's TagInfo on both Redmi Note 8 Pro: this can not even read the tag. That app says "Unknown Manufacturer

With a second tag:
- MCT on both both Redmi Note 8 Pro: same difficulties to connect and read
- NXP's TagInfo on both Redmi Note 8 Pro: this app can read them, but I see also differences between reads. Differences between reads on the same phone and between phones.

Either the type of tags I use are not meant to be read by these apps.
Or the Redmi Note 8 Pro has a fundamental issue.

I do not have access yet to another hardware to try.

ikarus23, 2020-05-27:
> I do not have access yet to another hardware to try.

That's a shame. Would be really interesting to find out if it the tag or the phone is causing the trouble. Maybe you can borrow a dive of a friend at some point?!

akhodaba, 2020-05-28:
Yes I will as soon as I can. Or even reading the tags on a windows PC if I can get hold of the right equipment.

ikarus23, 2020-08-15:
I will close this for now, because it looks like there is nothing I or MCT can do to fix this. But feel free to give feedback, if you learned something new. Still an interesting situation. ;)

closed 2020-08-15 by ikarus23