ikarus23/MifareClassicTool #297: Fix invalid access conditions?

nutinshell, 2020-06-03:
I accidentally wrote a CUID card 000000 access conditions using MCT, pm3 dump data now like:

```
      "AccessConditions": "00000000",
      "AccessConditionsText": {
        "block56": "rdAB wrAB incAB dectrAB",
        "block57": "rdAB wrAB incAB dectrAB",
        "block58": "rdAB wrAB incAB dectrAB",
        "block59": "rdAbyA rdCbyA rdBbyA wrBbyA",
        "UserData": "00"
```
is there any way to get it back to "FFFFFFFFFFFFFF078069FFFFFFFFFFFF" ?

Thanks.

ikarus23, 2020-06-08:
Some CUID tags can be saved using the PM3 `cwipe` command. Try it.

ikarus23, 2020-06-08:
Also, writing with `csetblk` should work.

nutinshell, 2020-06-09:
Hi, no luck:

```
[usb] pm3 --> hf mf cwipe
#db# wupC1 error
[!] ⚠️  Retry block[0]...
#db# wupC1 error
[!] ⚠️  Retry block[0]...
#db# wupC1 error
[!] ⚠️  Retry block[0]...
[!!] 🚨 Error setting block[0]: -1
[!!] 🚨 Can't wipe card. error=-1
[usb] pm3 --> hf mf csetblk 63 FFFFFFFFFFFFFF078069FFFFFFFFFFFF w
--block number:63 data:FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
#db# wupC1 error
[!!] 🚨 Can't write block. error=-1
```

ikarus23, 2020-06-11:
Looks like the the card is not compatible with the special wipe commands. I don't know of other ways to recover a card. It is possible the card is broken forever.

ikarus23, 2020-08-15:
I'm closing this. Sorry for your loss ;)

closed 2020-08-15 by ikarus23