ikarus23/MifareClassicTool #301: Factory format don't work well

weegeeluigi999, 2020-08-10:
I've tried to format a mifare 1k card with changeable uid on my honor 20 lite because there was an error and I thought it would repair it or at least delete it but it just don't want to work. the error is on the sector 0 the keyB is 12 "-" in case of 12 "0". what should I do please ?

ikarus23, 2020-08-11:
Not sure if I understand you correctly. In order to reset sector 0 you need key B because it has the write permissions and you can't use key B because MCT does not detect the key B as `000000000000`. Correct?

ikarus23, 2020-08-11:
It sounds like the Honor 20 Lite has this issue: https://github.com/ikarus23/MifareClassicTool/issues/66. I don't think there is anything I can do. Do you have another device? Or an USB reader?

weegeeluigi999, 2020-08-11:
Yes correct

weegeeluigi999, 2020-08-11:
No I don't but you mean that if I change the device it could work right?

ikarus23, 2020-08-11:
It sounds like a bug some NFC enabled phones have. If key B is known (e.g. `000000000000`) and the access bits are not damaged, then it should work. If the access bits are damaged (e.g. also `000000`), then the tag might be broken beyond repair. If it is a magic tag (e.g. gen1a) then you might be able to repair it with a USB RFID reader or the proxmark3

weegeeluigi999, 2020-08-11:
I think this is a magic tag is it? Model is Gen 2 CUID t4u (as I see)

weegeeluigi999, 2020-08-11:
So you mean that when keyB don't work it just crash and may be irreparable, right?

weegeeluigi999, 2020-08-11:
The error is : Sector 0, all blocks : key with write access (B) not known

weegeeluigi999, 2020-08-11:
Nevermind I don't have magic card the tag is broken I'll let you close this issue because they are not in use for me anymore

ikarus23, 2020-08-11:
Ok. However, if your phone is affected by #66 then you might be able to recover the tag.

closed 2020-08-11 by ikarus23