ikarus23/MifareClassicTool #304: Import .dump file directly?

jixiangyuan, 2020-09-01:
Hi,
The latest update was amazing, but I just realised that we can only import .mct .json .bin/.mfd and .eml. file through the port. Although it works when I choose the .json option when I try to import my dump file, but it still be nice if we can have the option to import .dump file directly. Thank you.

ikarus23, 2020-09-01:
Hi and thank you. I did not realize there is a `.dump` format. Is that from the original Proxmark3 repository? Or what tool generates these files?

You said the format is the same as the Proxmark3 RRG repository's `.json` file? So I could just display `.json/.dump`. Users might get less confused.

jixiangyuan, 2020-09-01:
> Hi and thank you. I did not realize there is a `.dump` format. Is that from the original Proxmark3 repository? Or what tool generates these files?
> 
> You said the format is the same as the Proxmark3 RRG repository's `.json` file? So I could just display `.json/.dump`. Users might get less confused.

Yes, I get the .dump file from a proxmark 3 GUI, also the Chameleon on android by RFID group generate a .dump file after sniffing. And sorry, the format should work as the same as the .bin/.dump option, so display it as .json/.dump definitely clear the confusion.

jixiangyuan, 2020-09-01:
> Hi and thank you. I did not realize there is a `.dump` format. Is that from the original Proxmark3 repository? Or what tool generates these files?
> 
> You said the format is the same as the Proxmark3 RRG repository's `.json` file? So I could just display `.json/.dump`. Users might get less confused.

`https://drive.google.com/file/d/1N9KSnOd7g8m-2cAC4K29-4Fn-9T5r2bZ/view?usp=sharing` I will give you this as a .dump file sample. And I also noticed that the app will crush everytime I try to import a dump file for a 4K card, is this a bug? or only 1K cards are supported?

ikarus23, 2020-09-01:
Thank you for the dump. Yes, it is indeed just the same as `.bin`. Therefore I will change the import to `.bin/.mfd/.dump`.

Importing a 4k dump should work. Could you provide me with a sample that crashes too?

jixiangyuan, 2020-09-01:
> Thank you for the dump. Yes, it is indeed just the same as `.bin`. Therefore I will change the import to `.bin/.mfd/.dump`.
> 
> Importing a 4k dump should work. Could you provide me with a sample that crashes too?

No worries. That's weird, the last time I tired all three 4K dumps crashed the app, but it all work fine now that I try it. Maybe it just some random error that occured. I will get back to you if I find any crash related to that in the furture.

Here is the file I tested `https://drive.google.com/file/d/1H9DOh_JZzO2QhDN4B9kU0Ok0uqVGkHVZ/view?usp=drivesdk`

Thank you.

ikarus23, 2020-09-01:
Interesting. If you manage to crash the app again, just send the crash report and open another issue. Maybe we can fix it.

I've added the clarification that bin = mfd = dump to the app. I will be in the next release. Thanks for reporting!

closed 2020-09-01 by ikarus23