ikarus23/MifareClassicTool #305: Saving files in Android 11

ikarus23, 2020-09-03:
https://twitter.com/evait_security/status/1301412304072978433
>We hit a few issues running under Android 11. Saving new files isn't possible anymore..only updating existing ones...any idea? Cheers.

ikarus23, 2020-09-03:
It looks like some permissions and/or structures for file access on the external storage must be updated. https://developer.android.com/preview/privacy/storage

ikarus23, 2020-09-03:
Ok, this new storage model will change a lot! The concept of storing files to the external storage and leaving them there after the app got uninstalled is no longer possible. So there a a few things to do:

* (Blacklist Android 11 until changes have been implemented)
* Use only the internal storage.
* Migrate data from the external storage to the internal storage.
* Remove the option to use the external storage.
* Make the export directory choosable (a higher Android API level is most likely required).
* Add dedicated dialog to warn users of data loss upon app uninstallation.
* Update the offline help & info section to reflect the changes.

blacklist-arcc, 2020-09-03:
Damn. This looks like a lot of work. is There any workaround/hotfix/patch/mokey-patch/whatever to get this workin in minimal?

ikarus23, 2020-09-03:
Yeah... I which I noticed this earlier. But as a hobbyist I'm not reading through all that Android developer news. I'm on holidays soon and not sure, if I will be able to fix this before that.

So far I haven't come across an dirty patch. I guess I have to do it the hard way. Hopefully I can skip the migration part. Maybe just use the new import tool to get the old files from the external storage to the internal storage.

ikarus23, 2020-09-03:
@ArchangeGabriel Oh, never thought of that. You might be right. I've already added the flag in the release, but targeted SDK 30 instead of 29. Let's try it!

ikarus23, 2020-09-03:
I've build a [testing version](http://tests.icaria.de/MifareClassicTool-3.0.1-test1.apk) with @ArchangeGabriel suggested temporary fix. People with Android 11: please try it. Especially creating new dump/keys.

blacklist-arcc, 2020-09-03:
nice, works like a charm - saving keys/dumps! Thanks for your time, have a nice day

ikarus23, 2020-09-03:
Great! Thanks for testing. I will release a new version with this quick fix until the real solution is implemented.

ikarus23, 2020-10-27:
I guess that's it. With 9afc6cb1ea17d411912a55cb9fc3c7f98298a91d everything should be compliant to the new Android storage model. The permission to read/write the external storage is no longer needed. It has been a challenging journey. Hopefully everything works as expected.

If you want, you can try a [test version of MCT (4.0.0)](http://tests.icaria.de/MifareClassicTool-4.0.0-test1.apk).

closed 2020-10-27 by ikarus23

blacklist-arcc, 2020-10-27:
Done testing. Works as expected - thanks for your great support!