ikarus23/MifareClassicTool #306: Access condition of last sector when factory formatting

jixiangyuan, 2020-09-03:
Hi, I just realised that when I try to do a factory format on a CUID Card (or Chinese magic card, gen 2), the last sector's access condition will become FF0780BC. I know the last last byte of the access condition is knid of useless, but I still don't understand why it is not FF078069. Thank you.

ikarus23, 2020-09-03:
Hi! The 0x69 is in there because it refers to non-personalized cards according to [AN10787 - MIFARE Application Directory (MAD)](https://www.nxp.com/docs/en/application-note/AN10787.pdf). The "General-purpose byte (GPB)" might be different when a MAD is in use.

However, I'm not exactly sure, why the last GPB of a 1k MIFARE Classic card is 0xBC. I've never looked into MAD deep enough (it's on the todo list https://github.com/ikarus23/MifareClassicTool/issues/30). I've just set it to 0xBC because factory new original tags by NXP had it set to 0xBC.

It's on 0xBC on other examples as well: http://nfc-tools.org/index.php/Libnfc:nfc-mfclassic.

jixiangyuan, 2020-09-03:
> Hi! The 0x69 is in there because it refers to non-personalized cards according to [AN10787 - MIFARE Application Directory (MAD)](https://www.nxp.com/docs/en/application-note/AN10787.pdf). The [...]

I actually never own any new NXP tags (since its NUID is not writeable), so I can't really check that myself. From what I saw [MIFARE CLASSIC 1K/4K USER MANUAL Release 1.1.0](https://www.google.com/url?sa=t&source=web&rct=j&url=https://shop.sonmicro.com/Downloads/MIFARECLASSIC-UM.pdf&ved=2ahUKEwiJ3ZSCks3rAhWQFIgKHdbSBGIQFjAKegQIBRAB&usg=AOvVaw2tvONHw50oQ-4uPCvlGSDR) P.9, the last bit is for "user data", so it's kind of a strange choice to see 0xBC here (actually the 0x69 doesn't make any sense here either, now that I think about it, maybe its there to distinguish from Key B). Anyways, not important, maybe give us an option to chooe that if you have time. Thanks :)

ikarus23, 2020-09-03:
Hehe, sure. But no promises.