ikarus23/MifareClassicTool #307: Error after update 3.0.2

longtrung94, 2020-09-05:
Helle everyone, sorry my bad english

I use note 10 and magic tag to clone UID
*Before update 3.0.2
Open app MCT. You tab the magic card MCT will show up UID of this card.

Tools -> Clone UID -> write 8 number of UID to clone- > Calculate block 0 and Clone UID -> Tab magic card -> Status log show succes no error -> Rescan to validate success -> successfully clone UID. Now if you tab magic card again, MCT will show UID. I can reuse this card to clone UID many times.

*After update 3.0.2
After clone UID successfull, I tab the magic card again, nothing happen, MCT not read and show UID of the magic card, this card can't reuse to clone UID, the computer still read UID of this card but the phone unrecognizable. I lost 2 magic card. The phone can read other card, but 2 card I clone successfull is can't.

Can you help me pls.

longtrung94, 2020-09-06:
I retry MCT 2.3.1, It work perfect with other magic card. But 2 magic card I use with 3.0.2 is not woking with phone and computer still read that magic card

ikarus23, 2020-09-12:
Thanks for reporting. It is indeed is an "bug" in the app. The problem is, that the rest of block 0 is wrong. It changes the ATQA of the tag to an invalid value. I'm sorry. I will fix this as fast as possible.

ikarus23, 2020-09-12:
Info by @doegox via Proxmark3 Discord:
>I tried MCT against a mfc 4k direct-write, tools / cloneUID. By default it had rest of block0 = 1D80184200040111001810 which gave an ATQA = 8018, incompatible with anticol, so I had quite a hard time to recover that tag with the help of patched pm3. Did I do sth wrong or should there be some check / other default in the tool ?

ikarus23, 2020-09-12:
Fixed with release of version 3.0.3.

closed 2020-09-12 by ikarus23

doegox, 2020-09-12:
FTR what I referred to as patched pm3 is now possible to achieve with the new `hf 14a config` Proxmark3/RRG command, so if you soft-bricked a tag with old MCT version, you can recover it with a Proxmark3.

ikarus23, 2020-09-13:
> FTR what I referred to as patched pm3 is now possible to achieve with the new `hf 14a config` Proxmark3/RRG command, so if you soft-bricked a tag with old MCT version, you can recover it with a Proxmark3.

This is great! Thank you so much for providing a way to fix broken tags!

ikarus23, 2020-09-13:
Example commands to recover soft-blocked tags with a Proxmark3: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md#proxmark3-commands-1