ikarus23/MifareClassicTool #309: Android open door

JulienRe, 2020-09-11:
Hello,

Is it possible to clone my badge on my phone, save it on my android phone and open my door with my phone ?

Thank you.

longtrung94, 2020-09-12:
You can do it, add new card on your locker, enable NFC on your phone, add phone as card. Enjoy it

JulienRe, 2020-09-12:
> You can do it, add new card on your locker, enable NFC on your phone, add phone as card. Enjoy it
> P/s only work with 13.56mhz card. Make sure screen unlocked alway on when you tab
Thank you for your answer. I wll try it.

JulienRe, 2020-09-12:
> You can do it, add new card on your locker, enable NFC on your phone, add phone as card. Enjoy it
> P/s only work with 13.56mhz card. Make sure screen unlocked alway on when you tab

Can you write here the application needed to do that please ? Thank you.

longtrung94, 2020-09-12:
I dont need any app, just on NFC

JulienRe, 2020-09-12:
With MiFare Classic Tool, I have "READ TAG" of my Vigik door Badge and I have saved it (TagDump.mct). How to open my door with my Phone ? What I need to do after ?

JulienRe, 2020-09-12:
Can we speak on Discord ? https://discord.gg/8zH3yC

longtrung94, 2020-09-12:
Can you add new door tag? When you add new tag, use your phone as new tag (enable NFC, phone's screen is on)
I am using my phone as member card at work

JulienRe, 2020-09-12:
Ahhhh okay I see. Thank you.

ikarus23, 2020-09-12:
Well this sounds like it is not related to MCT. It sounds like you are talking about a specific product. Since it seams everything is cleared up, I will close this issue for now.

On a side note: No, you can not simulate/emulate a MIFARE Classic card with your phone. At least not on a unrooted and unpatched phone.

Out of curiosity: What do you mean by "add phone as card"? How do you do that with an original Android and no extra app? I never saw such a function

closed 2020-09-12 by ikarus23

JulienRe, 2020-09-13:
I think it is not possible to open a Door with phone.