ikarus23/MifareClassicTool #310: Tag info

Bulte01, 2020-09-11:
Hello,

I have a card that I need to copy.

The original card has a 7 byte cl2 UID. and the card where I want to copy to has an 4 byte UID.
Is it possible to convert the 4 byte card to an7 byte?

Kind regards

longtrung94, 2020-09-12:
You need a USB card reader on PC, open notepad and tab this card near card reader, it will show UID on Decimal, convert to Hex 4 byte

Bulte01, 2020-09-12:
Thanks for the answer! 

I have a usb card reader but when I tab the card near the reader nothing will happens.
With the Mifarewindows tool I can read the card but not convert it.

longtrung94, 2020-09-12:
[image]
Do you have this?

Bulte01, 2020-09-12:
Not exactly the same but something similar.

[image]

longtrung94, 2020-09-12:
On my reader, tab the card, beep sound on, notead will show UID 
[image]

Bulte01, 2020-09-12:
> On my reader, tab the card, beep sound on, notead will show UID
> [image]

Ah oke mine does not do that.

But I already know the uid of my cards.

But the original card is 7 bytes and the card where I want to clone to is 4 bytes so i get a bcc error block 0 is not valid

longtrung94, 2020-09-12:
Give me 10 number UID of your card

Bulte01, 2020-09-12:
> Give me 10 number UID of your card

04452c5a015580

longtrung94, 2020-09-12:
Sorry, with your UID "04452c5a015580" I can't help.
I also have 1 card with the same number when read with phone 
[image]
,  but when using my card reader (as picture) it will show 10 numbers in the notepad. then I switch to hex code and write to magic card (finish clone)

ikarus23, 2020-09-12:
There is a lot of confusion here. So first of:
>Is it possible to convert the 4 byte card to an 7 byte?

No. You need a 7 byte UID changeable card. This has nothing to to with converting hexadecimal to a decimal UID.

There are 7 byte UID cards. Some of them are even compatible with MCT. [This one](https://de.aliexpress.com/item/32830423749.html) might work. These cards can be used to create a perfect clone.

closed 2020-10-16 by ikarus23