ikarus23/MifareClassicTool #312: Sector 0 Block 0 usually take 2-3 times to write

jixiangyuan, 2020-09-17:
Hi, I just found that the sector 0, block 0 of an Chinese Magic Card Gen2 usually can only be written with second or third try, and this happened to all my dump files, is this a bug or due to some protocal of android that can't be fixed? I will post a video to show what happenes when I tried. Thank you.

jixiangyuan, 2020-09-17:
Here is the video:

https://photos.app.goo.gl/fcKiGy4oVhndqwFM6

jixiangyuan, 2020-09-17:
Here's the dump files I used in the video:

https://drive.google.com/folderview?id=1DPC3CigSX7AXkHqqEwEjydZA3xusRgLu

ikarus23, 2020-09-24:
Interesting. Never head this issue. Do you have it with all Gen2 tags? Do you have another phone to test if it behaves the same?

ikarus23, 2020-09-24:
Also, try to remove the tag from the reader field before writing to block 0 and see if this reduces the attemps.

jixiangyuan, 2020-09-25:
> Interesting. Never head this issue. Do you have it with all Gen2 tags? Do you have another phone to test if it behaves the same?

Yeah this happens to all my Gen 2 tags from different manufactures, I used my Samsung note20ultra originally, and I tried using my Redmi K20pro, both devices cannot write sector 0 block 0 at once.

jixiangyuan, 2020-09-25:
> Also, try to remove the tag from the reader field before writing to block 0 and see if this reduces the attemps.

I tried this as you said, but still, it will take more than one time to write. Hmm, its a weird problem.

ikarus23, 2020-09-26:
Well, debugging this might get more complicated. My approach would be to sniff the communication between phone and tag with a proxmark3. There must be difference between an error and a successful write. If not, It is an internal issue with the tag you bought. There are so many different types out there... Or maybe just an antenna issue... I honestly don't know.

jixiangyuan, 2020-09-26:
> Well, debugging this might get more complicated. My approach would be to sniff the communication between phone and tag with a proxmark3. There must be difference between an error and a successful [...]

Sure, I will check that using my proxmark when I have time, but I doubt that this is a tag issue, as my tags brought from different seller experience the same issue. It's alright, not a big problem, can live with this if it can't be fixed.

ikarus23, 2020-10-01:
> I will check that using my proxmark when I have time

Great! Thank you. I will close this issue for now, because there is nothing I can do. If you find something new during your research, please report it here and feel free to reopen this issue.

closed 2020-10-01 by ikarus23