ikarus23/MifareClassicTool #318: Add a key file generator tool for know algorithems

ikarus23, 2020-10-31:
There are several known key generation/derivation algorithms out the to calculate the keys for a MIFARE Classic tag. Some of them can be found here: https://github.com/RfidResearchGroup/proxmark3/blob/master/common/generator.c. It would be great to have tool in which the user can input parameters (e.g. the toy/algo, the UID, etc.) and it generates a key file with the correct keys.

The0ne94, 2020-11-10:
Hi sir, what do you need for this project? Thank you for all your really Great job.

ikarus23, 2020-11-14:
I can do the UI stuff my self, but if you want you can port some of the known generators to java. I'm thinking like one function per product/generator. So something like:

```
String[] generateVingLock(String uid) {
    ...
    return keys;
}

String[] generateYaleDoorman(String uid) {
    ...
    return keys;
}

...
```
Be aware that there are helper functions in [Common.java](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/Common.java) like `hex2Bytes()` or `bytes2Hex()`.

dtd442, 2024-09-08:
that's interesting, how implement algos