ikarus23/MifareClassicTool #320: Info

mondlicht-und-sterne, 2020-11-03:
https://f-droid.org/en/docs/All_About_Descriptions_Graphics_and_Screenshots/

sterlingpollard, 2020-11-04:
GAY

LESBIAN

ikarus23, 2020-11-04:
I think, what you are trying to say to me with this link, is to update the information in the F-Droid store, right?
Well, you are right. This has to be done at some point. I will look into it. But I'm happy if someone want's  to help out on this. I will provide new screenshot soon (they are also needed for Google Play).

mondlicht-und-sterne, 2020-11-04:
Yes i mean screenshots ;)

ikarus23, 2020-11-06:
I've updated the general YAML and the `summary.txt` and the `description.txt` (https://gitlab.com/ikarus23/fdroiddata/-/commits/master). But because I'm not very familiar with the F-Droid structure, I have some questions:
* Are the changes that I made correct (e.g. encoding `&` as `&amp;`)?
* I deleted the `Description` from the YAML since I added the description.txt. But will it work for other languages?
* Where should I place the screenshots? Not in the fdroiddata, right?
* In builds, the folders `logo` and `tools` seem to be deleted. However the folder `logo` was replaced by `pics`. What should I do?
* Is there an easy way to visually check my changes?

I would really appreciate if someone could help me out on this. It takes me a lot of time to dig into F-Droids inner workings. Maybe somebody is already familiar with all this and could implement the changes quickly.

Bubu, 2020-11-07:
> Are the changes that I made correct (e.g. encoding & as &amp;)?

That should work, yes.

> I deleted the Description from the YAML since I added the description.txt. But will it work for other languages?

Not sure what the question here is. The fallback language is always english.

> Where should I place the screenshots? Not in the fdroiddata, right?

This should be places in your repo here, best together with the description and summary files. They'll only be picked up with a new release though.

> In builds, the folders logo and tools seem to be deleted. However the folder logo was replaced by pics. What should I do?

Submit a change to the metdata file please.

> Is there an easy way to visually check my changes?

Unfortunately no :(.

ikarus23, 2020-11-07:
Hi @Bubu! Thanks so much for your feedback.
> That should work, yes.

Great!

> Not sure what the question here is. The fallback language is always english.

Perfect. If en-us is picked as default if there is no summary.txt/description.txt found for another language, this should be fine.

> This should be places in your repo here, best together with the description and summary files. They'll only be picked up with a new release though.

Oh, ok, I get it now. I can move everything to this repo by just placing it at `root/metadata/...` I think I will do that.

mondlicht-und-sterne, 2020-11-07:
@ikarus23 it would be nice 👍

ikarus23, 2020-11-09:
Ok, I think I have everything together. So I've posted an issue to hopefully get this updated in F-Droid.
https://gitlab.com/fdroid/fdroiddata/-/issues/2213

ikarus23, 2020-11-09:
Everything is being updated. I think we are good to go.

closed 2020-11-09 by ikarus23