ikarus23/MifareClassicTool #326: My dump & key files are gone!

ikarus23, 2020-11-12:
**Your files are still there! Please read this!**
Updating from version < 4.0.0: If you updated this app from a version prior to 4.0.0 it might appear, that all dump and key files are gone. This is not the case. With version 4.0.0 MCT's file storage model has changed. You can get your files back by simply using the Import/Export Tool. The old files are stored on the external storage (SD card) in a folder named MifareClassicTool. Dump files are in the format .mct and key files are in the format .keys. It is possible to import multiple files at once.

This app stores data in the app-specific, internal storage. Files like a saved dump from a read-out tag can not be access by any other app or with a computer (USB). If you want to use them elsewhere, you can use the share button or MCT's Import/Export Tool.

**Be aware! Uninstalling this app will delete all files (dumps/keys) permanently.**

The regular version and the donate version of MCT can not share the same storage. Therefore each version has its own dump an key files.

Gideon03arti, 2021-06-27:
on

closed 2021-11-20 by ikarus23

Faroziogy5, 2023-02-21:
How this sistem process..