ikarus23/MifareClassicTool #327: Problem Dump in 4.0

fmaldonado2b, 2020-11-14:
L'application est très fonctionnelle, malheureusement la version 4.0 ne me permet plus de récupérer les fichiers de dump malgré la tentative d'import une erreur error incomplete or oversized dump est affichée...

[garage juin.zip](https://github.com/ikarus23/MifareClassicTool/files/5541406/garage.juin.zip)

ikarus23, 2020-11-14:
Hi!

There are dump files in the ZIP file. Just the default key files. So what are you trying to do? You can import them, but there is no point to it. Do you want to import your old dump files? Please read this carefully: https://github.com/ikarus23/MifareClassicTool/issues/326

fmaldonado2b, 2020-11-14:
Yes i want import my old dump files.Envoyé depuis mon téléphone Huawei-------- Message original --------De : Gerhard Klostermeier <notifications@github.com>Date : sam. 14 nov. 2020 à 21:39À : ikarus23/MifareClassicTool <MifareClassicTool@noreply.github.com>Cc : fmaldonado2b <franck.maldonado@laposte.net>, Author <author@noreply.github.com>Objet : Re: [ikarus23/MifareClassicTool] Problem Dump in 4.0  (#327)
Hi!
There are dump files in the ZIP file. Just the default key files. So what are you trying to do? You can import them, but there is no point to it. Do you want to import your old dump files? Please read this carefully: #326

ikarus23, 2020-11-14:
Well, then **[read this link](https://github.com/ikarus23/MifareClassicTool/issues/326)** and follow the instructions!

fmaldonado2b, 2020-11-15:
Bonjour, Le problème est le même quand je souhaite importer un dump mct il m indique une erreur, incomplète or oversize dump.FranckEnvoyé depuis mon téléphone Huawei-------- Message original --------De : Gerhard Klostermeier <notifications@github.com>Date : sam. 14 nov. 2020 à 21:52À : ikarus23/MifareClassicTool <MifareClassicTool@noreply.github.com>Cc : fmaldonado2b <franck.maldonado@laposte.net>, Author <author@noreply.github.com>Objet : Re: [ikarus23/MifareClassicTool] Problem Dump in 4.0  (#327)
Well, then read this link and follow the instructions!

ikarus23, 2020-11-15:
Sorry, but I can't help you this way. You have to be more specific! Provide me with the exact dump you want to import. Only then can I have a look what may cause this. Also, please be really specific in what you are doing. Or better, can you make a short video, showing what you are trying to do?

ikarus23, 2020-11-15:
To import a dump you have to press
Tools --> Import/Export/Convert Files --> Import Dump --> .mct (MIFARE Classic tool) --> select a dump file generated by MCT (no ZIP file) --> Open --> finished!

fmaldonado2b, 2020-11-15:
Bonjour,Je vous joins le fichier et toujours le même message d erreur , et la copie écran du message.Cordialement FranckEnvoyé depuis mon téléphone Huawei-------- Message original --------De : Gerhard Klostermeier <notifications@github.com>Date : dim. 15 nov. 2020 à 11:36À : ikarus23/MifareClassicTool <MifareClassicTool@noreply.github.com>Cc : fmaldonado2b <franck.maldonado@laposte.net>, Author <author@noreply.github.com>Objet : Re: [ikarus23/MifareClassicTool] Problem Dump in 4.0  (#327)
To import a dump you have to press
Tools --> Import/Export/Convert Files --> Import Dump --> .mct (MIFARE Classic tool) --> select a dump file generated by MCT (no ZIP file) --> Open --> finished!

fmaldonado2b, 2020-11-15:
Merci a priori les derniers dump fonctionnent, les plus anciens non.Donc problème résolu.MerciEnvoyé depuis mon téléphone Huawei-------- Message original --------De : Gerhard Klostermeier <notifications@github.com>Date : dim. 15 nov. 2020 à 11:36À : ikarus23/MifareClassicTool <MifareClassicTool@noreply.github.com>Cc : fmaldonado2b <franck.maldonado@laposte.net>, Author <author@noreply.github.com>Objet : Re: [ikarus23/MifareClassicTool] Problem Dump in 4.0  (#327)
To import a dump you have to press
Tools --> Import/Export/Convert Files --> Import Dump --> .mct (MIFARE Classic tool) --> select a dump file generated by MCT (no ZIP file) --> Open --> finished!

closed 2020-11-15 by ikarus23