ikarus23/MifareClassicTool #336: Support "Gen3" magic tags

ikarus23, 2021-01-30:
The new MIFARE Classic special/magic cards "Gen3" should be configurable using Android. The configuration commands are just issued using the APDU structure: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md#mifare-classic-apdu-aka-gen3

The main goal is to implement an extra Gen3 configuration tool. A great addition would be to have writing to block 0 of Gen3 seamlessly integrated into the rest of the application flow.

A way to detect Gen3 cards is to read block 0 without authentication. If this works it *should* be Gen3.

ikarus23, 2021-02-01:
@doegox:
> What I noticed is that using the APDU to change the UID doesn't change the block0 (which can still contain the previous UID), but using APDU to write block0 seems to change both block0 (with ATQA/SAK/BCC corrected) and UID.

ikarus23, 2021-02-01:
Step 1: I need to buy Gen3 tags!

ikarus23, 2021-05-22:
I got some tags. I did some implementation. It did change the UID but it also bricked my tag (partially). All very strange. The function is part of master (bd5d481e05e1eba00865e3b611a2f9045da1ff14) but it is not used/triggered for now.

Maybe someone is willing to contribute to testing at the risk of breaking gen3 tags? :)

Aendrew87, 2022-04-20:
hey,

just tried your modifications - yes, it modified the uid and bricked the card :)

The card is detected with the new uid (on android and linux (nfc-list)).
Your App is not able to read the card (got timeout) - mfoc returns "UNKNOWN_KEY" and "No sector encrypted with the default key has been found, exiting..".

This was the card i used:
https://www.amazon.com/dp/B07QM1GWL6?ref_=cm_sw_r_cp_ud_dp_DF1G8SNVF3R3B8DXXYSH

I had to modify one line to get it compiled
> writeGen3block0 = writeBlock0Gen3(data, key, useAsKeyB);

into

> writeGen3block0 = writeBlock0Gen3(data);

Aendrew87, 2022-04-23:
what i found while looking at the output of the app:

block 0 of original card:  
04 12 19 c3 cc 98 02 **00 64 8e** 85 94 41 10 38 07

block 0 the app tries to write:  
04 12 19 c3 cc 98 02 **08 44 00** 85 94 41 10 38 07

ikarus23, 2022-04-23:
Hi @Aendrew87! Thank you very much for testing and I'm sorry for you loss of a card. The behavior you're describing sound exactly the same as I experienced it.

Interesting catch with the difference in writing. How did you observe it? Where you able to read block 0 after it was written? Did you emulate a card and looked you for what MCT was trying to write?

The three bytes in question should be the ATQA and the SAK. The `08 44 00` seem to be the correct values. Somebody noticed that "ATQA and SAK bytes are automatically replaced by fixed values" on Gen3 tags when writing block0. But to my knowledge this is done by the card, and not Android/MCT. Also, this makes the original block look strange...

82ghost82, 2024-05-12:
It seems I have one of this gen3 cards and I confirm this is one of the strangest cards I had in my hands.
I'm sure it "was" a 7bytes uid 4k card. I was experimenting about how to write block0 with an acr122u without the PCSC Mifare software provided for these cards.
Following the proxmark commands I tried to write the whole block 0 with command `90  F0  CC  CC  10 <block0>`; so I provided a proper block0 for a 7 bytes card.
From that moment the card started acting strangely..I couldn't read the sectors because no key was found, and I never changed the default ffffffffffff keys. After running again the pcsc software I was able to read again the card but listen...now the card is a 4bytes uid 4k instead of 7 bytes uid...and...if I change the 4 bytes uid I cannot read anymore the card, because I found that probably the new uid that is written in block 0 is different than that read by the nfc reader. But I still can change the other things in block0. Note that I never blocked block0, nor used the lock command.
A proxmark3 is on its way to home, I need to try with it and see if I could restore this card to 7 bytes 4k, but I don't think it will be a success. No sense in all of this in my opinion.

ikarus23, 2024-05-13:
Interesting observations. Scrolling through https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md looks like there are special tags which allow changing between 4/7 bytes UID. Never had one of those. Since there are so many magic MIFARE Classic tags now, it's hard to keep track.

pergolafabio, 2025-01-03:
Does anyone know if 7bytes stickers also exist? I can only find fobs/cards ?

pergolafabio, 2025-02-15:
> I got some tags. I did some implementation. It did change the UID but it also bricked my tag (partially). All very strange. The function is part of master ([bd5d481](https://github.com/ikarus23/MifareClassicTool/commit/bd5d481e05e1eba00865e3b611a2f9045da1ff14)) but it is not used/triggered for now.
> 
> Maybe someone is willing to contribute to testing at the risk of breaking gen3 tags? :)

Hi, is there any progress on gen3 support? :-)

RainCat1998, 2025-05-05:
> It seems I have one of these gen3 cards and I confirm this is one of the strangest cards I had in my hands. I'm sure it "was" a 7bytes uid 4k card. I was experimenting about how to write block0 with an acr122u without the PCSC Mifare software provided for these cards. Following the proxmark commands I tried to write the whole block 0 with command `90 F0 CC CC 10 <block0>`; so I provided a proper block0 for a 7 bytes card. From that moment the card started acting strangely..I couldn't read the sectors because no key was found, and I never changed the default ffffffffffff keys. After running again the pcsc software I was able to read again the card (can't remember the exact steps but I played with the 4bytes uid fields..) but listen...now the card is a 4bytes uid 4k instead of 7 bytes uid...and...if I change the 4 bytes uid I cannot read anymore the card, because I found that probably the new uid that is written in block 0 is different than that read by the nfc reader. But I still can change the other things in block0. Note that I never blocked block0, nor used the lock command. A proxmark3 is on its way to home, I need to try with it and see if I could restore this card to 7 bytes 4k, but I don't think it will be a success. No sense in all of this in my opinion.

@82ghost82 I have a bold idea — is it possible that Gen3 magic cards are actually made from Gen4 magic cards by modifying the backdoor password?

> If the password is not the default, the tag might not be correctly identified by the latest Proxmark3 client (it might get mislabeled as MFC Gen2/CUID, Gen3/APDU, or NTAG21x Modifiable, depending on the configured UID/ATQA/SAK/ATS).
source: https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/magic_cards_notes.md

RainCat1998, 2025-05-05:
> It seems I have one of these gen3 cards and I confirm this is one of the strangest cards I had in my hands. I'm sure it "was" a 7bytes uid 4k card. I was experimenting about how to write block0 with [...]

Gen4 UMC card can change card type. It's possible you just triggered this behavior (or a bug related to it).