ikarus23/MifareClassicTool #342: Can't write block tag

charcmp, 2021-03-11:
Hi!
I'm trying to write a block in a mifare classic (magic) card 1k, the MCT app send the message that the write is correct, however when I read sector and block, the data has not been changed, I'm using a samsung galaxy s8. The key file is also correct. Key A and B are by default (FFFFFFFFFFFF).
Thank you in advanced.
Best regards.

ikarus23, 2021-03-17:
Hi! What block number are you trying to write to? What are the access conditions of the block (do you have the permission to write it)? Could you provide a dump of the sector in which the block is?

charcmp, 2021-04-05:
Hi Gerhard! Sorry for the delay in my response. I have permission to write,
the key A is correct. I think that the cause could be the tag, I was trying
to write in the rest of sectors and the card only allows me to write the
first time. If I try this operation in another card (from another vendor)
the write function works.

-- 
*Charo Campomanes Álvarez *| Ph.D. in Artificial Intelligence
@: ***@***.***

[image: CTIC]

*CTIC Centro Tecnológico / W3C España*
Parque Científico y Tecnológico de Gijón
C/ Ada Byron, 39  | 33203 Gijón  |  Asturias  |  España (Spain)
Tel: (+34) 984 291 212

Web <http://www.fundacionctic.org/>  |  Twitter
<https://twitter.com/fundacionctic>  |  LinkedIn
<https://www.linkedin.com/company/fundacion-ctic>  | Privacidad
<http://www.fundacionctic.org/privacidad>

ikarus23, 2021-04-17:
Very interesting. So the tag you have is one-time-writable? I never heared of such tags.

ikarus23, 2021-05-02:
Closing this for now, because it sounds like there is nothing I can do.

closed 2021-05-02 by ikarus23