ikarus23/MifareClassicTool #343: Writer dump error

yeriis, 2021-03-26:
What i have ti do if this error appear?(Version 2.3.1)
[image]
P.S:i have tried to use a recente version but it doesnt find me dump.

ikarus23, 2021-04-01:
First of all: Why use the very old and outdated version?
In more recent versions of MCT, you can import the dumps as described here: https://github.com/ikarus23/MifareClassicTool/issues/326
Now to your question: The dump is not a correct format. Like the error message says, each sector must have 4 lines (blocks) of data. You should check your dump file.

ikarus23, 2021-05-02:
Closed. Lack of feedback.

closed 2021-05-02 by ikarus23