ikarus23/MifareClassicTool #344: Speed problem

LeoFloresPCX, 2021-04-28:
Hia mates, How are You?

I have a speed problem:

I use MCT app with two phones: 1) A Moto One (p30 play) and 2) A LG G7 ThinQ.

When i read a Mifare card the keys's mapping lasts very different depending in which phone i use. All this is using the same card, same key file, etc.

In the Moto One the key's mapping lasts 2min 30s and in the LG G7 ThinQ lasts 4min 27s.

Why is this happening? The LG G7 ThinQ is supposed to be faster than the Moto One.

ikarus23, 2021-05-02:
Hi. Thank you for reporting this. Unfortunately, I have no idea what could be causing this. But it is not the first time that I hear of phones behaving differently when it comes to NFC. Could be a vendor thing.

Are you comfortable with Android Studio and some debugging? This way you could get some more insides.

ikarus23, 2021-12-28:
I guess this needs more testing. However, I do not have access to two phones that show this behavior. I will leave this issue open for a while. Maybe someone can check it out further.