ikarus23/MifareClassicTool #346: Ability to write 7 byte UID tag

W4635, 2021-05-18:
Hi,
I made a copy using MCT of a Mifare classic card CL2 with a 7 byte UID.
However I am unable to write the dump to a magic tag due to wrong BCC value.
Forcing it to write wrong values resulted in a broken chip.
Is it possible to write to the chip using CL2 command or some other way to write without breaking the chip?

ikarus23, 2021-05-22:
It should be possible to write CL2 tags with a 7 byte UID, as long as it is gen2 card.. Last time I checked it worked for me. What are the exact steps you do? Maybe I missed something.

W4635, 2021-05-22:
How I try to write 7byte tag:
- Write tag
- Write Dump (Clone)
-- Show Options
-- Advanced: Enable writing to manufacturer block
- Select Dump (7 byte dump)
-- All sectors selected
-- Put Gen 2 tag in front of phone NFC reader
-- Ok
- Select keys from dump to write
- - Start mapping and write dump
- ERROR: BCC of block 0 is not valid

I think it calculates the BCC value as if it was 4 bytes and then finds out its incorrect?
Because I made a 1 on 1 copy of Mifare classic card CL2 and it works just fine at the door acces reader.
If you calculate the first 4 bytes and then the BCC value and other values after, I think the app correctly states that values are incorrect for 4 byte but it does not take into account its 7 byte?
My knowledge of the difference between 4 byte and 7 byte is not that great.

ikarus23, 2021-05-23:
Thank you for the explanation. MCT should not calculate/check the BCC for 7 byte UID tags. Could be a bug. I will look into it.

Regrading the "Clone UID" tool: This should work. The soft brick was because of a mistake I made in version 3.0.0 to 3.0.2 (if I recall correctly). It should not brick in tags in the latest version.

ikarus23, 2021-05-23:
I've just tested it again with a 1k gen2 CL2 7 byte UID tag. I can write block 0. There only is a warning saying that my data in block 0 could be wrong, but it works.

W4635, 2021-05-23:
Thanks for your dedication and helping me out!
I am cloning a dump from a regular 1K CL2 7 byte UID tag to a 1k gen2 4 byte UID tag. From your message I understand that I have to clone 7 byte to a 7 byte magic (2nd gen) tag and can't clone 7 byte to 4 byte even if its 2nd gen magic?

ikarus23, 2021-05-23:
Ohhh, now I understand. Yes, Cloning a 7 byte UID to a 4 byte UID tag is not possible. The length of the UID can not be changed. Therefore you need the correct magic tag to make a 100% identical clone.

W4635, 2021-05-23:
Then the issue is solved! Thank you!

closed 2021-05-23 by W4635