ikarus23/MifareClassicTool #347: [Bug] MCT doesn't work on OnePlus One and OnePlus 3T

taivlam, 2021-05-20:
## Description

I believe the OnePlus One and the OnePlus 3T are compatible devices that work with MCT.

However, both devices are not functioning properly with MCT, both running LOS 18.1 for microG (Android 11).  I am not sure if this is due to MCT or something wrong upstream with the respective LOS builds.

The last time I tried MCT with both devices was in October of 2019, both running LineageOS 16.0.  Yet, two years later, in May 2021, MCT no longer functions properly with these two devices.  (Despite this, MCT seems to work perfectly fine on the OnePlus 5T.)

### OnePlus One

When I turn on NFC and scan any MIFARE Classic card, the NFC seems to occasionally flicker on and off - even though I tapped the Quick Tile to activate NFC.  (To be specific, I scanned a CharlieCard.)

Initially, MCT will detect the card:
[image]

However, when I tap on "READ TAG", MCT will "quit" out on me by saying the NFC isn't turned on (even though it is):
[image]

MCT isn't always consistent on the OnePlus One - it tends to cycle non-periodically, with not much of a pattern of either going to the first error or second error.

### OnePlus 3T

When I use a CharlieCard on the OnePlus 3T, MCT will always tell me that the card is not supported - even though MCT correctly identifies the card as a MIFARE Classic 1k card from NXP:
[image]

ikarus23, 2021-05-22:
Hi. Thanks for the feedback. There were always trouble with at least the OnePlus One, if I remember correctly. Interesting to hear that it worked at some point with some version of LineageOS.

Regarding OnePlus One:
This is really strange. I can't even think of a way in which this should occur. especially by pressing the Read Tag button. I don't have a OnePlus One device to reproduce it. Do you feel comfortable with Android Studio and the debugger? :)

Regarding OnePlus 3T:
It does tell you that your device does not support the card. Not that MCT is not supporting the MIFARE Classic Charlie card. There are a lot of devices which have NFC but do not support the MIFARE Classic technology. There is nothing this app can do about it. It's just a hardware/driver/license thing.

taivlam, 2021-06-12:
## OnePlus One

I didn't know that the OnePlus One empirically had issues with MCT.  (I only attempted to pull use MCT on the OnePlus One because I wanted to be "scientific"/"reproducible" with instructions I was writing.)  It's been so long since LOS 16.0 came out, so I'm not sure what happened between then and now.

Although I'm unfamiliar with Android Studio and the debugger, I'm willing to work it out.  (If the overall experience using the debugger is more extensible and/or reliable than creating a bug report ZIP on the OnePlus One itself, then I'm all for it!)

## OnePlus 3T

Alright, you may be correct; IIRC, I never used MCT on the OP3T to obtain the keys (with exact slot positions) - not the OPO.  (Unrelated: however, I believe I managed to load the keys onto the OP3T and make Metrodroid work with the Charlie Card/MIFARE Classic 2 years ago.)  Back then, the OP3T might have shown an error message yet still worked; but it's been so long, so I can't say for sure.

ikarus23, 2021-06-20:
You are right. The OnePlus 3T should work as well. At least it did in the past. That is what users reported to me so I added it to the [list of compatible devices](https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md). Would be very interesting to see what have changed. Because I think I did not change something specific in the app. Maybe something on the Android/firmware changed.

Have you tried other MIFARE Classic compatible apps? Like [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite)? If they work well (you can specify keys), MCT should also work.

taivlam, 2021-06-24:
I'll get back to you after I download NXP TagInfo and see if the OP3T's NFC can function properly (or not).

wc7086, 2021-07-30:
I tested multiple NFC software on OS from Android11, but none of them can read and write CUID cards normally. The device I use is oneplus 3.

wc7086, 2021-08-07:
But I don't have any problems with Android 10.

ikarus23, 2021-11-20:
@taivlam Any news on testing NFC apps on your devices?

@wc7086 This topic/issue is not about CUID tags. There are several different tags out there which could cause trouble. For example, gen1a tags will not work regardless of the Android version or device. However, it seams like you are saying MCT and NFC apps in general did work on Android 10 on your OnePlus 3 and not with Andorid 11. Is this right? Did you test several apps?

taivlam, 2021-12-16:
I've been all over the place, but now I should be able to start testing (by this weekend at the latest).

Usually, I'm not comfortable installing proprietary apps on my phones.  However, since I no longer use either of these devices and they no longer have any important data, I'm ok with a proprietary app if it means getting to the bottom of this.

I can also use my Pixel 4a as a "control" device, since it is a supported device and I install it on an isolated secondary profile on GrapheneOS.

(Random side question: is the Pixel 5a supported by MCT?  I haven't seen any mention of some of the newer Google Pixels: the 4, 5, 5a, or 6.)

ikarus23, 2021-12-28:
Hi @taivlam. Would be great to find out more, but only if you have the time and feel comfortable with it. And there is a good chance that in the end we could not do anything about it, because it is a driver/ROM issue.

Sorry, I don't know if the Pixel 5a supports MCT. I'm still on my 3a and did not hear anything from a 5a user.

taivlam, 2022-01-16:
@ikarus23 I've managed to test on the Pixel 4a and OP5T, so I know what a proper NXP Taginfo scan should look like.  (Strangely, an empty dialog box with only one option box for "Ok" keeps popping up, as if the NFC is turn off or something is wrong with the Pixel 4a.  Despite this, the Pixel 4a works fine.)

Should I save & provide the information displayed by the NXP Taginfo app, or do the results below sufficiently indicate that the OP3T and/or the OPO wouldn't work anyways?

### OP3T

BLUF:  NXP Taginfo were silently incomplete, despite no explicit warning from the app when the scan was complete.

The NXP Taginfo app made sounds as if the NFC scan proceeded with no errors.  However, there is no NDEF data under the NDEF tab.  Also, under Full Scan tab, the OP3T is lacking `android.nfc.tech.MifareClassic` and `android.nfc.tech.NdefFormatable` under the "Android technology information" section; and the "Memory content" section is completely absent.

(I'd like to note that on the first initial scan or scans on the OP3T, NXP Taginfo gave a warning/error that the device was not compatible or something among those lines.  However, this error never came up again.)

### OPO

BLUF: NXP Taginfo scans were also incomplete in a similar silent manner to the OP3T, but required moving the OPO in very strange manner to even obtain a scan.

I didn't really expect much, but the OPO behaved exactly like I've reported before.  If I tried to scan any CharlieCard/Mifare Classic card, then the OPO would either:
* do nothing, which was most of the time (while the NFC status color would flash/toggle between on and off in the Android 11 Quick Tile area); or
* try to bring me to the Android settings area to turn on NFC (because the NFC was perceived to be off), which was less likely.

(The initial NXP Taginfo scan may have also said the OPO was outright incompatible during the first scan/first few scans, but I haven't been able to make this warning/error message appear again.)

However, the OPO did manage to scan with a very unreliable method: slowly waving the phone back and forth between 0.5 inches to 2 inches away from the card while the card lay flat on the table.  I only was able to scan the card's NFC sensitive area if I held the OPO at an angle.  This angle was very awkward - the bottom of the phone had to be held 3 inches higher than the top and 1-2 inches to the right away from "center" position (if the bottom of the phone was closest to you).

The OPO's scan results were almost identical to the normal scan, except under the Full Scan tab the "Maximum transceive length" and "Default maximum transceive time-out" both had `[not available]` values; and again there was no "Memory content" section.

#### Random thought

Is it possible to permanently damage the NFC sensors by storing the OPO too closely to a [NFCKill](https://nfckill.com/) device?  (I have bought the NFCKill to destroy the NFC in old CharlieCards before shredding.)  At one point, I stored my NFCKill device a little too close to the OPO for comfort and I'm not sure if this fickle scanning OPO behavior is due to potential/possible NFC damage.

Even though I didn't really have as much hope for the OPO, I wanted to be thorough.

ThePinkUnicorn6, 2022-05-07:
Hi, I am having the same issue with my oneplus 3t, is there any chance of a fix being implemented or would that require modifying system files?

ccole38, 2022-05-17:
I have exactly the same issue on my oneplus one as taivlam did on 20th May . I'm running LineageOS 18.1 rooted with magisk and can try anything

Ewoodss, 2023-05-19:
The app also doesn't work on my oneplus 3t with lineage os 18.1

ikarus23, 2023-08-17:
Hi @taivlam and other. I totally missed out to get back to this top after @taivlam's epic analysis. Now, 1.5 years later, I found this issue again...

Unfortunately, it seems like "professional" apps like NXP's TagInfo struggle as well. `android.nfc.tech.MifareClassic` not being present when an MIFARE Classic tag is presented means, that the device does not support MIFARE Classic. I guess I will add a note to both devices in the `COMPATIBLE_DEVICES.md` saying something like "only with certain versions of ROMs". It seemed they worked at some point and stopped working with more recent versions. Nothing I can do there. Probably has something to do with the NFC drivers of LinageOS.

Thanks for all the research. I will close this now.

closed 2023-08-17 by ikarus23

taivlam, 2023-08-19:
I apologize for neglecting this issue (despite the prior occasional pings of activity), though I think @ikarus23 is basically correct w.r.t. closing this GH issue.  I was never able to figure a way to resolve the NFC issue, and I wouldn't wait around for this to be resolved in LOS, due to reasons already mentioned and described below.

I honestly haven't used the OnePlus One or 3T anymore, as in "put my SIM card into this smartphone" use.  The largest factor contributing to this is that both devices are "legacy devices" that will basically force these devices to stay on LineageOS 18.1/Android 11 until LOS 18.1 is no longer supported, according to the "Let’s talk about legacy devices…" section of the LineageOS 19 [announcement](https://lineageos.org/Changelog-26/).

Also, I started to wise up regarding OEM device firmware updates on Android, partly thanks to using GrapheneOS, starting with the Pixel 4a in early 2021.  Seeing as the newest OnePlus devices (since roughly 2019) no longer have the earliest advantage to obtaining custom Android OS releases anymore (between device release date and the first stable release of any aftermarket Android OS), and that the OPO and OP3T would've been out of OEM support for the OEM/Qualcomm SoC since 2017 and 2019 (respectively); I honestly can't recommend using these devices that have been EOL w.r.t. firmware updates in 2023 anymore.

Maybe, just maybe, these devices might be good to give to a younger person as a cheap &/or upcycled first smartphone to prevent creating immediate e-waste.  (You wouldn't be able to expect reasonable security levels when using a OPO or OP3T in 2023, but that's now a tangential topic.)  Also, maybe testing how an app works in a older but still currently supported Android OS version is viable, but that's about it.

Until recently, I hadn't been taking public transit on a somewhat regular basis, so I don't think I would've been passionately following up on this; and also I only recently started to use CharlieCards again.  (Though, from what I can tell, the Pixel a devices, including the Pixel 4a and 5a,  have been fine with NFC reading of MiFARE Classic cards.)

Lastly, for completeness, Issue #389 is probably a duplicate of this issue.  Both should be mentioned in commit 7a74994, but it seems the OP of the other GH issue didn't see this one.

ikarus23, 2023-08-19:
Thanks for the feedback!
Side note: If you're interested NFC, Security and the CharlieCard have a look at: https://www.youtube.com/watch?v=me6FyrDRlD4 ;)