ikarus23/MifareClassicTool #349: "No supported application for this NFC tag"

noasgr, 2021-07-22:
Hello,

I have been using the app successfully over the past months. 
Unfortunately, it´s been a while that every time I bring a tag close to my phone, I get the following message:
"No supported application for this NFC tag"
This happens regardless from having the app open or not. and the option "autostart the app if a tag is detected" end up in the same result..

The only hint here that I would like to add is that I have activated google pay on my device. However, I don´t know if this could affect somehow.

Any help would be much appreciated. If you need any further information, please let me know.

ikarus23, 2021-08-03:
Hi. MCT is not a (good) tool for tag detection. There is a filter in the app that tells Android it is only processes ISO 14443a tags. If you e.g. present a ISO 15693 tag MCT will not open regardless of the settings.

A good app for tag detection is [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite).

noasgr, 2021-08-03:
Hello and thanks for the answer. I just wanted to clarify that the very same tags used to work without any problem.
The only thing that changed since then was that I set up google pay on my phone (I don´t know if that could affect anyhow).
Thanks once again

ikarus23, 2021-08-03:
Very interesting. Does the tag detection work when MCT is running in the foreground?

noasgr, 2021-08-04:
No it doesn´t I have the app open, I place the tag close to the phone and I get the generic "No supported application for this NFC tag" message.

ikarus23, 2021-08-04:
Strange. So Android is detecting the tag but does not forward it to any other app. I guess the same message is showing even if you have tools like NXP TagInfo installed, right?

This sounds not normal. I guess normal NFC apps should still work if you have Google pay enabled. What device are you using?

noasgr, 2021-08-05:
I tried NXP TagInfo  and I didn´t receive that message.. it happens only with MCT..
My device is a Xiaomi Mi 10t lite.
Thanks

ikarus23, 2021-11-20:
Sorry for the late response. Is the issue still present? What do you mean by "I didn´t receive that message". Did NXP TagInfo work without any issues? Was is able to detect and read tags?

noasgr, 2021-11-22:
> Sorry for the late response. Is the issue still present? What do you mean by "I didn´t receive that message". Did NXP TagInfo work without any issues? Was is able to detect and read tags?

Hello,

This issue is no longer happening. It´s been a long time but to be honest I don´t recall taking any special actions (just made sure to update my phone) and it was fixed.

Many thanks

ikarus23, 2021-11-22:
Great to here! I'm happy as long as it works. :)

closed 2021-11-22 by ikarus23