ikarus23/MifareClassicTool #350: shows "NFC not enabled" after few seconds when starting mapping a mifare card

fogfog, 2021-08-05:
Hi, 
I've been using MCT for number of years, and since I changed my old galaxy S7 (on which MCT run perfectly) for a Galaxy Note 10 lite, I can't even start mapping and I got this error "NFC not enabled" popup message while it is enabled indeed.
Any way to fix that?

ikarus23, 2021-11-20:
Hi. Sorry for the late response. Never had this issue before. Unfortunately I don't have a Galaxy Note 10 Lite to reproduce this. Do you have this issue with other NFC apps? E.g. NFC TagInfo or NFC Tools?

ikarus23, 2021-12-28:
Any news on this? Have sou tried other apps?

ikarus23, 2022-01-19:
Interesting. So it should be a bug in this app. So to be precise: When you start MCT with NFC enabled you get a popup message to enable it? Or is it just a small toast message (bottom of the screen) that shows up at some point when you using the app?

ikarus23, 2022-04-23:
Hi! Any news on this?

ikarus23, 2023-08-18:
Closed due too lack of feedback.

Also, I was not able to reproduce it. Are there any other Galaxy Note 10 Lite users out there? Do you have the same issue?

Feel free to reopen this issue, if there are any news.