ikarus23/MifareClassicTool #352: Add a recycle bin for deleted dump file?

Warborn123, 2021-09-14:
I was just wondering could we get a feature like adding a recycle bin for deleted dump file? Becuase its pretty easy to delete a dump file accidently, and there is no way to get it back at the moment. Thank you!

ikarus23, 2021-11-20:
I don't know. Is it really that easy to do it by accident? You have to first press the menu button first and then select delete file. Implementing a recycle bin seems like a lot of effort for this corner case.

Maybe implementing a "are you sure you want to delete file xy?" dialog could help?

I will think about it and keep this issue open for now.

Warborn123, 2021-11-20:
> I don't know. Is it really that easy to do it by accident? You have to first press the menu button first and then select delete file. Implementing a recycle bin seems like a lot of effort for this [...]

Yeah, to be honest the dialog definitely helps, but sometimes for idiots like me, when a popup shows up, I will just click yes without having anything thought, so a recycle bin is definitely better, but you are the boss, do whatever you like :)

ikarus23, 2021-11-20:
Hehe, I get it. I'm one of those idiots as well. Even on the my Linux command line I delete files by putting them into the recycle bin.

>but you are the boss, do whatever you like

Well, it is open source. So anyone can contribute ;)