ikarus23/MifareClassicTool #353: Not writing to magic tag & not reading all data

Darnel-K, 2021-09-16:
Have been trying to write a dump to a magic tag using a oneplus 9 pro however it immediately says “tag lost while checking dump” and gives up every time, seems to be an issue on this phone specifically as a friend can write the same dump to the same tag with no problems but I don’t have any issue cloning the UID using the oneplus. When reading a tag it seems to miss data from sectors randomly resulting in me having to do multiple reads and manually piece together and fill in the blanks to create a full dump.

ikarus23, 2021-11-20:
It seams like the OnePlus 9 Pro suffers from the same issue as some other phones. The Nexus 5X behaves very similarly. It is possible that there are even more issues with your device. It's strange that always OnePlus devices have NFC problems.

However, there might be a fix. Not sure if it will fix everything for you, but the multiple manual reads and merging is not needed. Go to the settings, enable "Retry authentication" with something like 5 and maybe try the option "Auto reconnect".

ikarus23, 2021-12-28:
Any news on this? Did the settings help?

ikarus23, 2022-01-19:
I will close this for now. Feel free to reopen it if you have some news.

closed 2022-01-19 by ikarus23