ikarus23/MifareClassicTool #358: MCT not working after update to Android 11

dorcky, 2021-11-19:
Hello everyone. Just update my Cat S62 pro to Android 11 and MCT not work anymore. It say that this device does not support Mifare Classic. How so, because before A11 was working good. Anyone can help with this issue? Thank You!

ikarus23, 2021-11-20:
Strange. I guess there must be something wrong with the Android ROM of this device. What error do you get? And when? On app startup or when you trying to read a tag?

dorcky, 2021-11-20:
Right after i put a card on phone, not even try to read the card. On app startup.

ikarus23, 2021-11-20:
Could you open the app first (without a card close to your phone)? Is there an error? Or only if you place a card on your phone while the app is running in the foreground?

It probably is the latter. If the error occurs when a card is presented to the open app, then there is nothing I can do (as far as I know). The Android NFC API tells the app that one or both (the phone or the card) does not support MIFARE Classic. It seams like in your case this is a false positive introduced with the new Android version.

dorcky, 2021-11-20:
Only when i place the card on phone appear that message. The latest Android version is the main problem. It was a mistake to update to 11. Thank`s anyway!

ikarus23, 2022-01-19:
Another idea: Have you tried other apps to read a MIFARE Classic token? E.g. NFC TagInfo (by NXP). Use it with an empty MIFARE Classic card. This app should also be able to read the content as long as the sectors have default keys.

ikarus23, 2024-01-22:
Closed. Lack of feedback.
Feel free to reopen this issue if you have new information.

closed 2024-01-22 by ikarus23