ikarus23/MifareClassicTool #359: Error: There is no MIFARE Classic tag

Nathan-NL, 2021-11-22:
Since 4.0.3.

S20 4G running Android 11

ikarus23, 2021-11-22:
Thanks for reporting. I messed up. Sorry. I've just released version 4.0.4 with a fix.

ikarus23, 2021-11-22:
If you don't want to wait until Google Play updates, you can download it from here: https://www.icaria.de/mct/releases/4.0/

closed 2021-11-24 by ikarus23

raphllo17, 2022-08-31:
Even with the 4.0.4 version I alaways have the same issue on Android 11

RyanGibb, 2022-11-27:
I have the same issue on 4.0.4

rudddy, 2023-04-27:
Hello

I have the same issue with v4.0.4 and Google Pixel 7 Phone - Android 13 - 5.10.149-android13-4-00004-g0e5c0e91bbee-ab9475045

Thanks

ikarus23, 2023-05-21:
Hi @rudddy. Can you rule out, it is not an Mifare Classic tag? This app only works with Mifare Classic tags. If you try to read other tags, it will not work, because "There is no MIFARE Classic tag".

rudddy, 2023-05-21:
Hello

I don't who wich tag it's precisely

Thanks

Cordialement,

Yves Denan

valica, 2023-08-24:
Hello,
I'm using the 4.0.5 version (the latest at this moment and I still have the Toast telling Error: There is no MIFARE classic tag.
Do you know if it's really fixed. I've tried on Samsung S21 and still not fixed.
Could you tell me if I have something special to do to solve this issue, please ?
Kind regards

QuentiumYT, 2023-10-10:
Same here with a Google Pixel 7 :)