ikarus23/MifareClassicTool #361: Can't read sectors 0-7 after upgrade to Android 12

Immelstorn, 2021-12-11:
Month ago I used this app to read a card and created a dump of all sectors including sector 0.
Now I can't read first 8 sectors (0-8) of the *same* card. 
I wrote saved dump with sectors 0-7 to another empty magic card, it succeeded, but I can't read that card too.

[image]

The only thing that changed for this month was that I upgraded my phone to Android 12. I tried to install older versions of MCT, but it didn't help, and I lost my dump :(

I use Pixel 4XL if it matters.

Immelstorn, 2021-12-13:
And even more. I'm getting different results for the same card with the same phone/app version:
[image]
[image]
[image]
[image]

ikarus23, 2021-12-28:
This looks like I well known bug. It was present in Googles Nexus 5X but I haven't seen it since then. Too bad it seems to be back. Last time I checked, there was nothing I can do to fix it. So I implemented a workaround:

Go to the settings -> Retry authentication -> try a value like 5. 

This will slow down the mapping process. It will try up to X times to authenticate with the same key even the previous attempt did not work. But this is how the bug presents itself. Sometimes the Android NFC API says the authentication did not work even if the key was correct. Maybe it is some kind of race condition or an issue with specific tags...

Immelstorn, 2021-12-28:
Ok, thanks, trying this now. It takes a lot more time indeed. It is working for 40 minutes now and progress bar is still somewhere at the 30%.

ikarus23, 2021-12-28:
Oh wow. Yeah, the big key file can take long. If the keys are found don't forget to save the to a separate key file. You can do so by selecting the "Save keys" option from the menu of the dump editor.

Immelstorn, 2021-12-28:
Well... it took about 1 hour, and progress bar reached ~50% and then it said that it is finished, but first 8 sectors are still not found. I wonder why did it work for the very first time with Android 11. Is there so much difference  between 11 and 12 in this area?

ikarus23, 2021-12-28:
Strange. Maybe it lost the connection to the tag. But It's a pain to retry if it takes that long.

No, I don't know of big updates to the NFC API in Android in between 11 and 12.

ikarus23, 2022-01-19:
More and more people have issues like this after updating. But not always to Android 12. Sometimes people are getting this when updating to 11. It seems to only affect some devices. (Or maybe it is the card/token?!)

Anyways. I will close this for now. I'm sorry, but there is nothing I can do so far. I don't have access to a device with the same issue. Last time I had one, I could not do anything. It just was the API who reported an authentication failure sometimes, even if you used the correct key.

closed 2022-01-19 by ikarus23