ikarus23/MifareClassicTool #362: The keys cannot be edited in version 4.0.2 and above

MicroDuke, 2021-12-21:
I  do test the version 4.0.1 & 4.0.2 & 4.0.4.
The version 4.0.1 Can enter after ediiting, The picture is as follows

[image]
The version 4.0.2 & 4.0.4 can not enter after editing,The picture is as follows
[image]

This version does not allow screenshots

ikarus23, 2021-12-28:
Interesting. Did you try the different versions on the same Android device with the same Android version? So far I can not recreate this issue.

MicroDuke, 2021-12-29:
Yes, both of my phones are xiaomi: mix2 & Mi 11.
Their situation is exactly the same. 
The mix2  Andriod version is 9 ,MIUI is 12.0.1,Kernel is 4.4.153.
The Mi 11 Andriod version is 11 ,MIUI is 12.5.15,Kernel is 5.4.86-qgki. 
I have not tested other phones

MicroDuke, 2021-12-29:
The keyboard of version 4.02 and above cannot be selected, it is a Xiaomi security keyboard。

ikarus23, 2022-01-19:
I think I found the issue. Some text fields in this app are declared as `textVisiblePassword`. This is a quick hack to tell keyboards to show an extra row with numbers. This way it is easier to input hex. However, in your case the phone decides to use the "password keyboard", which does not allow the return key.

I guess I have to revert the `textVisiblePassword` definition from all input fields.

MicroDuke, 2022-01-20:
> I think I found the issue. Some text fields in this app are declared as `textVisiblePassword`. This is a quick hack to tell keyboards to show an extra row with numbers. This way it is easier to input hex. However, in your case the phone decides to use the "password keyboard", which does not allow the return key (as new line).
> 
> I guess I have to revert the `textVisiblePassword` definition from all input fields.

Thanks a lot.
I will test the new release version & report test report.

ikarus23, 2022-01-23:
I compiled a testing version. Feel free to try it: https://tests.icaria.de/MifareClassicTool-4.0.4-test1.apk

MicroDuke, 2022-01-24:
> I compiled a testing version. Feel free to try it: https://tests.icaria.de/MifareClassicTool-4.0.4-test1.apk (Edit: fixed link.)

Yes,I do test it .
It is work.
Thanks.

ikarus23, 2022-01-24:
Thanks for testing. The next version will include the fix.

closed 2022-01-24 by ikarus23