ikarus23/MifareClassicTool #364: Can't read key A of sectors 9

zlqd123, 2021-12-31:
The key A of sectors 9 read by mct was none. It should be FFFFFFFFFFFF.
[image]

[image]

MCT 4.0.4 from google play
CUID card
Xiaomi 10pro MIUI13 21.12.27

zlqd123, 2021-12-31:
The second picture should be this.
[image]

zlqd123, 2021-12-31:
MCT can't write the key A of sectors 9 either.

ikarus23, 2022-01-01:
It just looks like you don't know key A for this card.

zlqd123, 2022-01-02:
> 看起来你不知道这张卡的钥匙A。

I've got the full dump file and a key file. It could be read before.
Maybe it's Andriod12.

ikarus23, 2022-01-03:
Could be. There are other people having issues with Android 12. Please have a look at https://github.com/ikarus23/MifareClassicTool/issues/361#issuecomment-1001992143. It might help you,

ikarus23, 2022-01-19:
Have you tried one of the solutions?
It seems like more and more people have this issue after updating Android...

ikarus23, 2022-04-23:
I will close this for now. Feel free to reopen it if you have more information about the topic.

closed 2022-04-23 by ikarus23