ikarus23/MifareClassicTool #368: hex to datetime

iitaichi, 2022-01-07:
Need some help to determine how is this date/time being encoded 

Here are some examples (known date only):

+Sector: 0
22725A3B3108040003F1BE2C02F0131D
00000000000000000000000000000000
00000000000000000DAE2B412B9F0200
------------7F078869FFFFFFFFFFFF

+Sector: 0
53BB61F67F080400024A319516F9931D
00000000000000000000000000000000
00000000000000000C952A212C3F0200
------------7F078869FFFFFFFFFFFF

   2B5F0200 -> 31/10/2021
   2B9F0200 -> 31/12/2021
   2C3F0200 -> 31/01/2022

 Idon't understand how this datetime format works.

ikarus23, 2022-01-08:
I can not spot it by a quick look. But are you sure these 4 bytes encode a date? How did you come to that conclusion?  Is it a hotel system? Are there online or offline readers? Are the rest of the blocks empty? Do you have the dump of the same card before and after changing the date? Do you already know the meaning of the other 4 bytes?

ikarus23, 2022-01-19:
Since an issue tracker is no good place to ask for this kind of questions, I will close this now.

Feel free to  post more information here if you have some. Otherwise I can recommend the [Proxmark3 Forum](http://www.proxmark.org/forum/index.php) or the RFID Hacking Discord.

closed 2022-01-19 by ikarus23