ikarus23/MifareClassicTool #373: Probleme

Skyguesh, 2022-01-28:
Je n'arrive pas à créé mon badge malgré la bonne fréquence de badge original et copie avec block 0. Tout ce passe bien mais a la fin un message dit: ERROR: Some error occured while writing. (This might be bad.)
J'essaie de copier avec Xiaomi 9T.
J'ai essayé plusieurs fois, mais toujours le même problème. Si j'arrive à résoudre je mettrai 5 étoiles

Skyguesh, 2022-01-28:
Version 4.0.4 (j'ai déjà désinstaller et réinstaller. J'ai eu toujours le problème, j'ai changé de badge copie et toujours le même problème...

ikarus23, 2022-01-29:
Hi. Need some more information here. Are you using a block 0 writable tag (gen2 aka. CUID)? If so, where did you get it from? Do you have a link?

Skyguesh, 2022-01-29:
Bonjour, merci de m'aider.
Voici le lien ou j'ai acheté mes badges 
Pouvez vous me dire ci c'est des bon badge ?
https://a.aliexpress.com/_m0KFUfA

ikarus23, 2022-01-30:
Yes, looks like the tags are not compatible. From the documentation:
>The first block of the first sector of an original MIFARE Classic tag is read-only i.e. not writable. But there are special MIFARE Classic tags that support writing to the manufacturer block with a simple write command. This App is able to write to such tags and can therefore create fully correct clones.
However, some special tags require a special command sequence to put them into the state where writing to the manufacturer block is possible. These tags will not work.

What you need are MCT compatible tags. They go by the name "CUID" or "Gen 2". Something like this should work: https://de.aliexpress.com/item/32956735195.html

Skyguesh, 2022-01-31:
Merci, je vais donc essayer ça.

closed 2022-01-31 by ikarus23