ikarus23/MifareClassicTool #378: invalid acces conditions or dead sector

dawaman, 2022-03-18:
Hello

When I copy RFID tag I have this below error.

[image]

I have a dump. I write tag with theses options:
Clone
with option seleted:
Uses theses access  conditions for all sectors
Advanced: Enable writing to manufacturer block

Do you know why I have this error?

ikarus23, 2022-03-29:
You have this error because of the reason right there in the error message: "Key with write access (B) not known." MIFARE Classic tags have two keys per sector (key A and B). Every sector has access conditions (access rights). A common configuration is "reading with key A, writing with key B". It seems like you have key A and were able to read the tag. But it seems like do not have key B to write the tag.

If you are willing to post the dump here, I can have a closer look.

dawaman, 2022-03-30:
Thanks for your reply. 

Please find my dump.
[UID_651284EC_2020-10-18_12-25-28.zip](https://github.com/ikarus23/MifareClassicTool/files/8378698/UID_651284EC_2020-10-18_12-25-28.zip)

dawaman, 2022-03-30:
below, the dump from the tag
[image]

dawaman, 2022-03-31:
thanks for your reply.

Is it possible to retrieve the key B?

ikarus23, 2022-04-15:
Yes, in most cases this is possible. Due to some weaknesses in MIFARE Classic, you can retrieve all the keys (A and B) of a tag with tools like the [Proxmark3](http://www.proxmark.org/) or normal RFID-Readers and some special software ([mfcuk](https://github.com/nfc-tools/mfcuk), [mfoc](https://github.com/nfc-tools/mfoc)).

closed 2022-04-15 by ikarus23