ikarus23/MifareClassicTool #379: Error there is no MIFARE Classic tag

Jasperion, 2022-03-18:
Samsung Galaxy S8 here.
I'm trying to read a tag, but I can't éven get to that as it gives meg this error message instantly.
As far As I know Galaxy S8 shoud be compatible.

ikarus23, 2022-03-29:
Hi. Yes, as far as I know the Galaxy S8 started to work from Android 8. Have you had this issue before? Was there an update for the S8? On what version are you now? Also, are you sure you are presenting a MIFARE classic tag? Have you tried apps like NXP's TagInfo?

Jasperion, 2022-03-30:
Hi! Tbh I just started trying this out, so I'm not sure if there was an issue beforehand. I am running android 12, but since then have tried to use a J5 and an S10, all of which yielded the same result. I have tried the app you mentioned, but no result whatsoever. 
I feel like it doesn't even read the tag, because I get the error message instantly after pressing the "start mapping and read tag" button. (It was the

Jasperion, 2022-03-31:
No response whatsoever - so I guess it doesn't detect the tag

ikarus23, 2022-04-15:
Yes. In that case the tag is most likely not a MIFARE Classic tag. Most likely not even a NFC tag. There are a lot of RFID tags that are not compatible with phones. For example, some of them use a completely different frequency (125KHz instead of 13.56MHz).

closed 2022-04-15 by ikarus23