ikarus23/MifareClassicTool #381: Can you use NfcAdapter.enableReaderMode() to implement this app?

zhpncu, 2022-04-04:
Hi ，

Can you use NfcAdapter.enableReaderMode() to implement this app?

Because this method already exists after android 4.4, this method is better for reading the card

https://developer.android.com/reference/android/nfc/NfcAdapter
public void enableReaderMode (Activity activity, 
                NfcAdapter.ReaderCallback callback, 
                int flags, 
                Bundle extras)

ikarus23, 2022-04-15:
That is a good idea. I always wanted to use this function, but when I started MCT it was compatible all the way down to Android 2.3 which had no `enableReaderMode()`. Since the current minimum is Android 4.4 there should be no issue in using it.

zhpncu, 2022-04-17:
Yes, but now there are few phones below Android 4.4, the latest to Android 12, I still recommend you see if you can add this function, if it is compatible with Android 4.4 before, you can also use android sdk level to judge it.

Yuki1536, 2024-11-13:
Without calling that method the phone may not enter Reader Mode at the time touching another phone with card mode.

The opposite phone enter the Reader Mode and my phone at MCT homepage forced to close and enter the card swiping page,