ikarus23/MifareClassicTool #386: Ability to emulate Mifare card

Footsiefat, 2022-05-10:
This is the only app that supports mifare cards and I was wondering if it was possible to emulate them from my (rooted) phone too? It'd be very nice for thing such as having a copy of my access cards on my phone and id even be willing to pay for such a feature :+1:

AkechiShiro, 2022-05-10:
I don't think it's possible with the app at the moment, but I haven't done a very thorough/extensive use of it, maybe with another app it might be doable, but doing such would also probably depend on the hardware used for NFC inside your phone, I'm not too sure.

A device more suited for this would be a [Proxmark3](https://github.com/RfidResearchGroup/proxmark3), it is specific hardware that you can connect to your phone via USB I believe and use it for such a purpose, card emulation, card cloning too and much more.

It costs about 60-80$, be sure to take the 512KB memory (sticker 512M) if you go for it, as the more space you have, the more the firmware will have features.

I would be interested in only using my phone to do such a task too, but I'm not sure if it is doable, hoping someone knowledgeable can hop into this issue and enlighten us both.

ArchangeGabriel, 2022-05-11:
See https://github.com/ikarus23/MifareClassicTool/issues/153, https://github.com/ikarus23/MifareClassicTool/issues/156.

ArchangeGabriel, 2022-05-11:
There is also the [Chameleon Mini](https://shop.kasper.it/chameleonmini/170/chameleonmini-revg-color?c=5), a bit cheaper than a Proxmark 3 that from my experience costs more like 300$.

AkechiShiro, 2022-05-11:
@ArchangeGabriel I guess it depends on from where you're buying it ? I'm in France (European Union) and on Amazon, there are [Proxmark 3 512KB listed for 61€](https://www.amazon.fr/Proxmark3-Develop-Proxmark-Lecteur-copieur/dp/B09F3H7SD7/) or more: 
[image]

Chameleon Mini are all sold out on lots of stores for me but may be cheaper as well, 3 times cheaper probably, however the Chameleon Mini also has a lot less features but still has card emulation I believe.

Footsiefat, 2022-05-12:
> @ArchangeGabriel I guess it depends on from where you're buying it ? I'm in France (European Union) and on Amazon, there are [Proxmark 3 512KB listed for [...]

Sorry @AkechiShiro but as cool as that is, and I appreciate you finding that for me. I wanted to see if I could use my phone by itself because of the pure portability and convenience factor.

AkechiShiro, 2022-05-12:
I totally understand your point, its just isn't simple and there really is no support for full Mifare card emulation I think on phones, there are very few possibilities on a phone for card emulation so far, as been mentioned in prior comments/issues, with specific devices there are issues and support looks a bit random, it's not as complete as this solution hence why I proposed this one to you.

I would have liked it that it could have been done with a phone only as well, so that I could save some money and for convenience purposes but it looks like not to be very feature wise for now.

Footsiefat, 2022-05-12:
Understood and I appreciate you taking your time to respond and help me out with this, I guess ill close this issue now to help make sure too many don't build up.

closed 2022-05-12 by Footsiefat

AkechiShiro, 2022-06-21:
If such a project see the light, could anyone ping it to me, I'd be happy to contribute to it, looks pretty cool even if it's just using xposed/root Android device it's better than nothing.

DavidBerdik, 2022-06-21:
> I believe it would be supported but of course undertaking this would take a very large time and effort.

Oh yes for sure. I would not be opposed to experimenting with it myself especially since I have a Proxmark which I could use for testing, but at the moment, I have too many other side projects that need wrapped up and can't invest the time needed to figure out how the puzzle pieces fit together.

> If such a project see the light, could anyone ping it to me, I'd be happy to contribute to it, looks pretty cool even if it's just using xposed/root Android device it's better than nothing.

If I ever pick it up, I'll be sure to let you know, but at least for now, that is not going to happen. You could always try picking it up yourself! 🙂

I found [this app on F-Droid](https://f-droid.org/packages/com.vsmartcard.acardemulator/) that has implemented some form of NFC card emulation. I've opened up a new issue on this app's GitHub repo (https://github.com/frankmorgner/vsmartcard/issues/226). Hopefully the developer will be able (and willing) to either implement it or provide some advice on what implementing it would entail.