ikarus23/MifareClassicTool #387: MCT not finding any cards on OnePlus 6T

borisbsv, 2022-05-23:
Hey! I'm new to RFID/NFC so it's most likely human error, but wanted to try and clear things up here.

Running OnePlus 6T on Android 11 (OxygenOS). When trying to read a mifare chip with MCT I find nothing - it doesn't even detect a chip. (NFC is on on the device and I regularly use it, mostly for pairing/paying). I'd be happy to provide any further details that can help!

ikarus23, 2022-06-20:
Hi. Sorry for the late response. The OnePlus 6T should be working. At least it was reported to me as working. Are you a 100% sure your tag is MIFARE Classic? There are RFID tags that look exactly the same but are using 125kHz frequency. Phones can not talk to these tags.

Have you tried other apps? E.g. TagInfo by NXP is a good one.

ikarus23, 2022-09-28:
Closed. Lack of feedback. If there any relevant news, feel free to reopen the issue.

closed 2022-09-28 by ikarus23