ikarus23/MifareClassicTool #389: Device does (not) support Mifare Classic

mmm4m5m, 2022-06-18:
Maybe the message is wrong. Or checking for Mifare Classic support is incorrect.

1. Metrodroid application report that phone have support for Mifare Classic
"You can verify MIFARE Classic support in Metrodroid's about screen."
https://github.com/metrodroid/metrodroid/wiki/Cracking-keys

2. As far I could find and read in the code, my phone do not have 'bcm' or 'bcrm' files:
`$ adb shell find /dev /system | grep -i 'bcm\|bcrm'`

> hasMifareClassicSupport: ...this function will check if there are files like "/dev/bcm2079x-i2c" or "/system/lib/libnfc-bcrm*"

https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/Common.java

[image]

ikarus23, 2022-06-20:
Hi. Detecting if a device supports MIFARE Classic is hard. Without presenting a MIFARE Classic tag to your device you can not be sure. While I don't think my detection mechanism might be right all the time, I don't think the one of Metrodroid is either. Have you checked other apps like TagInfo by NXP?

ikarus23, 2022-06-20:
Also, what device are you using?

mmm4m5m, 2022-06-25:
@ikarus23
Device is OnePlus3T with LineageOS.
I guess the NFC files/drives are vendor specific - not from LineageOS.

I do not know much about NFC (yet).
Is Tag.techList not enough for "yes/no" Mifare support?
Above message (not supported) is shown after tap NFC card.

I am looking mostly at open source examples.
If I find good example I will let you know.
Here from TagInfo by NXP - the same NFC card - city transport card:

[image]

[image]

[image]

ikarus23, 2022-09-05:
> @ikarus23 Device is OnePlus3T with LineageOS. I guess the NFC files/drives are vendor specific - not from LineageOS.

Yes, the drivers in LineageOS often differ from the original ones because of legal issues (licensing)

> I do not know much about NFC (yet). Is Tag.techList not enough for "yes/no" Mifare support? Above message (not supported) is shown after tap NFC card.

It is a simple yes/no list, but it only gets filled with data when a tag is touched. If MIFARE Classic is not in the techList, you don't know why. Could be the device not supporting it or the tag not being MIFARE Classic.

> I am looking mostly at open source examples. If I find good example I will let you know. Here from TagInfo by NXP - the same NFC card - city transport card:

The "Full Scan" tab of NFC TagInfo is the important one. Here you will see data, if the device was capable of reading out your tag.

closed 2022-09-28 by ikarus23