ikarus23/MifareClassicTool #392: MCT can't read sector with different A and B keys

JackCarterSmith, 2022-08-16:
I want to read a tag with:
- default sector with FFFFFFFFFFFF A-B keys, 
- sectors 4 to 8 with keys A0A1A2A3A4A5 for A and 458767340675 for B.

I specified a key file with following entries:
- FFFFFFFFFFFF 
- A0A1A2A3A4A5 
- 458767340675 

I can manually read sectors with FFFF.. keys, but I'm not able to do the same with the others differents A-B keys sectors.

Anything I mistaken?

JackCarterSmith, 2022-08-16:
Yes, I used fake keys for the example.

ikarus23, 2022-09-05:
Hmm, not sure. What device are you using? Are you sure you have the correct keys? Have you verified this with another reader (e.g. ACR 122u USB RFID reader). Do you have access to another phone and tried it from there as well?

JackCarterSmith, 2022-09-06:
I tried on a Blackview 9800 Pro, normally compatible for mifare tech.
When using a proxmark mk3 I doesn't have issues to dump it (that also a confirmation for the keys working good).
I don't have spare phone to test unfortunately...

ikarus23, 2022-09-10:
Really strange. Have you tried other apps like NXP's TagInfo. You can add custom keys there and check by doing a "Full Scan".

JackCarterSmith, 2022-09-12:
I have tried with NXP's taginfo app. I think the reader is "uncompatible", it's read false datas from tag and doesn't seem to found good keys, like if tag or reader return unexpected result.

I don't know what chip is used in BV9800 Pro. Maybe this model is also uncompatible?

ikarus23, 2022-09-28:
Even more strange. Reading false data seams really annoying. Reading no data and throwing an error would be way better to debug. But yes, I think it might be your device (or its firmware/drivers), unfortunately.

If you ever come across another phone, just try it. Until then, I will close this.

closed 2022-09-28 by ikarus23