ikarus23/MifareClassicTool #394: Can't get keyA which is 000000000000

heyzzk, 2022-09-06:
### I read my tag with MCT and found that I couldn't get the KeyA of sector 3,8,9,11,12,13,14,15, which seem to be 000000000000.
### Let's take sector 3 for example, it's like this.
[image]

### Buf if I read the tag with a usb nfc reader. The KeyAs that MCT can't get are 000000000000.
[image]

### The keys I used are like below,
  FFFFFFFFFFFF
  A0A1A2A3A4A5
  D3F7D3F7D3F7
  000000000000

### In order to figure out why, I add a Log.i here ,
[image]
### then read the tag and get the log like below, so you can see that even though I use the right keyA 000000000000, I still can't pass the authentication.
  09-06 23:41:00.551 10416 11411 I  MCReader: sector=0, try key=FFFFFFFFFFFF, auth=true
  09-06 23:41:00.573 10416 11411 I  MCReader: sector=1, try key=FFFFFFFFFFFF, auth=true
  09-06 23:41:00.600 10416 11411 I  MCReader: sector=2, try key=FFFFFFFFFFFF, auth=true
  09-06 23:41:00.640 10416 11411 I  MCReader: sector=3, try key=FFFFFFFFFFFF, auth=false
  09-06 23:41:00.682 10416 11411 I  MCReader: sector=3, try key=D3F7D3F7D3F7, auth=false
  09-06 23:41:00.714 10416 11411 I  MCReader: **sector=3, try key=000000000000, auth=false**
  09-06 23:41:00.746 10416 11411 I  MCReader: sector=3, try key=A0A1A2A3A4A5, auth=false
  09-06 23:41:00.758 10416 11411 I  MCReader: sector=4, try key=FFFFFFFFFFFF, auth=true

### and finally I get the keys like this,
  09-06 23:41:01.873 10416 10416 I  KeyMapCreator: 0 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.877 10416 10416 I  KeyMapCreator: 1 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.880 10416 10416 I  KeyMapCreator: 2 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.882 10416 10416 I  KeyMapCreator: 3 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.885 10416 10416 I  KeyMapCreator: 4 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.888 10416 10416 I  KeyMapCreator: 5 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.891 10416 10416 I  KeyMapCreator: 6 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.894 10416 10416 I  KeyMapCreator: 7 : FFFFFFFFFFFF,FFFFFFFFFFFF
  09-06 23:41:01.895 10416 10416 I  KeyMapCreator: 8 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.897 10416 10416 I  KeyMapCreator: 9 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.898 10416 10416 I  KeyMapCreator: 10 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.899 10416 10416 I  KeyMapCreator: 11 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.901 10416 10416 I  KeyMapCreator: 12 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.902 10416 10416 I  KeyMapCreator: 13 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.903 10416 10416 I  KeyMapCreator: 14 : ------------,FFFFFFFFFFFF
  09-06 23:41:01.905 10416 10416 I  KeyMapCreator: 15 : ------------,FFFFFFFFFFFF

### Could you please give me some clue? Thanks.

heyzzk, 2022-09-07:
I made some mistakes. It's not 000000000000.

heyzzk, 2022-09-09:
Seems the host have changed this sectors to not default value，a way to prevent from copying? Therefore the MCT failed get the keyA.

ikarus23, 2022-09-28:
Hi, yes, key `000000000000` was only used by the other program because there is no marker for *unknown data* if binary dumps.

I don't know what your *host* is. Do you mean the reader of the system that normally reads the card? Because yes, there are systems that change the key based on an algorithm.

I will close this for now. If there is something related to MCT, feel free to reopen the issue.

closed 2022-09-28 by ikarus23