ikarus23/MifareClassicTool #395: Add light theme

xmha97, 2022-09-10:
Please add the light theme.

ikarus23, 2022-09-28:
NEVER! Hacking must be done always in dark mode. :D

All joking aside, providing a proper light and dark theme depending on the system settings seems the right thing to have. But I'm not strong when it comes to UX. Also, to be fair, not very motivated. Maybe someone is willing to have closer look and provide a PR.

karageee, 2023-06-26:
light mode is cursed

ikarus23, 2023-12-29:
To follow the system settings, it is basically enough to set the `styles.xml` to:
```
<resources>
    <style name="AppTheme" parent="Theme.AppCompat.DayNight" />
</resources>
```
However, there are still some fixes needed. The "blue" color of MCT should be the themes accent color. And some text (e.g. HexToAscii tool) do not follow the color scheme for some reason...

ikarus23, 2024-01-02:
Uff, I just wanted to quickly add this feature since using AppCompat provides a light and dark theme.

It had been hours and days of work! I had to learn so much about Androids theme system. Also, I needed to fix a lot of theme related bugs and had to test this on multiple versions. Even back to Android 4.4.

The light theme does not look great. The color palette is still optimized for the dark theme, but I will stop here for now. This "simple" feature took way too many of my resources. :)

Choosing themes is possible as of 075456993c9299c474ad11bcd07f2315fda13184 and will be part of the next release

closed 2024-01-02 by ikarus23