ikarus23/MifareClassicTool #396: How to generate valid sector 0 data by the already UID number?

f2sky, 2022-09-10:
As shown in the picture below, only sectors 0-5 of a card are read.

If a UID number is: AE4CA2C1, how do I generate valid data for sector 0 block 0 (i.e. the line of data marked in red in the picture)?

Trying to use the clone UID function, but it always fails.

[image]

ikarus23, 2022-09-28:
Hi! Not sure if I understand correctly. There seem to be a language barrier. 

1) What do you mean by that? Do you mean the "Clone UID" tool? What was the error message? For me it looks like you are using a incompatible tag (only on MFIARE Classic Gen2/CUID tags block 0 can be written by MCT). Have a look at: https://github.com/ikarus23/MifareClassicTool#general-information
2) It looks like your tag is a MIFARE Classic Mini. *Mini* tags only have 5 sectors. You can check using NXPs TagInfo app.

f2sky, 2022-09-29:
My tags work properly with the "Write Dump(Clone)",but not "Clone UID".
Please see the following screenshots.

[image]

[image]

ikarus23, 2022-10-03:
Yes, like I said, I does look like you are using an incompatible tag. Just like the error message says. You can not write block 0 (the UID) of a normal tag. You need a special tag for that. These tags are often called MIFARE Classic Magic gen2/CUID.

ikarus23, 2022-11-02:
Closing this for now. If you have more questions about this, feel free to reopen it.

closed 2022-11-02 by ikarus23