ikarus23/MifareClassicTool #397: Need help about initialize the NFC

zhxst, 2022-09-11:
Hey, this MCT tool is very cool. I'm learning to create a flutter plugin for simple use.
Currently I just learing form your source code for simple usage like read and write card.

The card is 4 bytes magic card, and works well with MCT.
But I use my demo code, it success to detect the card, but fail on authenticate the key.
And here is the STRANGE thing. After I use the MCT to detect the card, my demo can suddenly successfully auth the key and write the block. It works until I reboot the phone.

So, what am I missing before authenticating the key? 
Or is there something special to initialize when detect the card?
Any advice will be very helpful. Thank you.

zhxst, 2022-09-11:
It's a plugin, so no interface. I use `nfcAdapter.enableReaderMode`.
When tag detected I use `patchTag`, then `MCReader.get(tag)`, then use `reader.connect()` and then `reader.writeBlock`.

zhxst, 2022-09-13:
After tried many times, it comes out that, I need to detect a card outside my app any where. Then I can auth the card inside the app. So weird.

zhxst, 2022-09-16:
Found answer on [stackoverflow](https://stackoverflow.com/questions/72003781/mifare-classic-authentication-failure-on-some-devices-until-the-tag-has-been-rea).
So it's because `enableReaderMode` is somehow different from `enableForegrounDispatch`, which cause `patchTag` fail.

zhxst, 2022-09-17:
Me again. The major problem is actually missing `Intent Filter` about `ACTION_TECH_DISCOVERED`.
So both `enableReaderMode` and `enableForegrounDispatch` works.
Still, the problem about last comment on stack overflow is maybe exist.
So I decide to change to `enableForegrounDispatch` for possibly better Compatibility.

ikarus23, 2022-09-28:
Hi. Finally found some time to look at MCT's issue list. What a journey you had. Yes, lots of pitfalls. And yes, `enableReaderMode` is something completely different than foreground dispatching.

Not sure if there is anything left I can help you with. If so, say so.

Best of luck!

closed 2022-10-22 by ikarus23